mavlink/qgroundcontrol #7445: Android: Large number of crashes from QGCActivity.nativeUpdateAvailableJoysticks

DonLakeFlyer, 2019-05-15:
This is being reporting as 100s of crashes on the Google Play store stable build:

Today, 9:58 AM on app version 352000
Samsung Galaxy Note5 (noblelte), Android 7.0
Report 1

java.lang.UnsatisfiedLinkError: 
  at org.mavlink.qgroundcontrol.QGCActivity.nativeUpdateAvailableJoysticks (Native Method)
  at org.mavlink.qgroundcontrol.QGCActivity.access$800 (QGCActivity.java:65)
  at org.mavlink.qgroundcontrol.QGCActivity$3.onReceive (QGCActivity.java:169)
  at android.app.LoadedApk$ReceiverDispatcher$Args.run (LoadedApk.java:1185)
  at android.os.Handler.handleCallback (Handler.java:751)
  at android.os.Handler.dispatchMessage (Handler.java:95)
  at android.os.Looper.loop (Looper.java:154)
  at android.app.ActivityThread.main (ActivityThread.java:6682)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run (ZygoteInit.java:1520)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1410)

DonLakeFlyer, 2019-05-15:
@MatejFranceskin It's critical that you look at this one. It's blowing up all over the place on people using Stable build.

MatejFranceskin, 2019-05-16:
I am looking at it but the problem is that I cannot reproduce it on my devices. It is ok with or without joystick.
Do you have any more information? It could be dependent on Android version.

MatejFranceskin, 2019-05-16:
When did this problems start? Immediately after it was included in the build or later?
Did we change target SDK version recently?

DonLakeFlyer, 2019-05-16:
Looks like all over the place:
[image]

Do you have access to Google Play Dev Store? Probably should. Daniel should be able to get it for you. Then you can see crash reports directly.

sjd69, 2019-05-16:
I've replicated this issue using a Note 8 (Qualcomm) as well as a S9+ (Qualcomm).

In my case it happens when opening QGroundControl with a DJI drone (using RosettaDrone) already connected. Starting QGroundControl prior to connecting the drone resolves the issue seemingly permanently and allows me start and restart QGroundControl freely in any order. Issue will appear again on a fresh install.

```
    --------- beginning of main
2019-05-16 11:41:39.731 12119-12119/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: org.mavlink.qgroundcontrol, PID: 12119
    java.lang.UnsatisfiedLinkError: No implementation found for void org.mavlink.qgroundcontrol.QGCActivity.nativeUpdateAvailableJoysticks() (tried Java_org_mavlink_qgroundcontrol_QGCActivity_nativeUpdateAvailableJoysticks and Java_org_mavlink_qgroundcontrol_QGCActivity_nativeUpdateAvailableJoysticks__)
        at org.mavlink.qgroundcontrol.QGCActivity.nativeUpdateAvailableJoysticks(Native Method)
        at org.mavlink.qgroundcontrol.QGCActivity.access$800(QGCActivity.java:65)
        at org.mavlink.qgroundcontrol.QGCActivity$3.onReceive(QGCActivity.java:169)
        at android.app.LoadedApk$ReceiverDispatcher$Args.lambda$-android_app_LoadedApk$ReceiverDispatcher$Args_52226(LoadedApk.java:1319)
        at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.$m$0(Unknown Source:4)
        at android.app.-$Lambda$FilBqgnXJrN9Mgyks1XHeAxzSTk.run(Unknown Source:0)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6940)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
```

DonLakeFlyer, 2019-05-16:
@sjd69 Can you try the QGC Daily build from the store? I believe that has the fix in it now.

sjd69, 2019-05-16:
@DonLakeFlyer It appears to fix the issue on both devices.

DonLakeFlyer, 2019-05-16:
@sjd69 Awesome, thanks so much for testing.

DonLakeFlyer, 2019-06-15:
This needs to be in Stable branch as well

DonLakeFlyer, 2019-06-27:
@MatejFranceskin Can you move this to Stable_V3.5.1?

closed 2019-07-02 by DonLakeFlyer