mavlink/qgroundcontrol #7453: MAVLinkFTP for CameraManager & ArduPilot

olliw42, 2019-05-16:
@dogmaphobic
I ping you upon suggestion by @julianoes. 

This is neither a bug nor issue nor request report. It is an effort to seek your support for working towards a more complete camera and component handling, as outlined in this issue: #1133 Proposal for Improving User-fiendliness of Component Handling.

In the recent MAVLink devcall it was concluded to approach this in a step-by-step prototyping way, and a first crucial step would be to make it possible that the camera definition file is grabbed from a STorM32 gimbal controller (which is also a camera controller) using MAVLinkFTP. If I'm not mistaken, @julianoes wants to approach it from the DronecodeSDK side.

As much as I've read, QGC has already a quite rich MAVLinkFTP support. However, some few little pieces seem to be still missing to complete this. It seems that QGC has MAVLinkFTP disabled for ArduPilot. Since I'm still on ArduPilot it would be great if that could be enabled. Please note also https://github.com/mavlink/mavlink/issues/1133#issuecomment-492541303 (which forwards to https://github.com/ArduPilot/ardupilot/pull/11183). Also, according to https://github.com/mavlink/mavlink/issues/1133#issuecomment-493004692, QGC does not yet use MAVLinkFTP to fetch the camera definition file.

An additional point to consider is that, for resource reasons, a component like the STorM32 could not support the full-blown MAVLinkFTP protocol, i.e., QGC should do the trick with using only a very minimal MAVLinkFTP, which essentially consists of only the Read mechanism, but NACKs all other opcodes (for more details see https://github.com/mavlink/mavlink/issues/1133#issuecomment-492870605).

So, this issue here is about asking you if you would be willing to work towards these points. I would be most happy and eager to do the STorM32 side.

I in fact have already added MAVLinkFTP code to it, which on the drawing board should work, but I have no means to test it. And I frankly admit that my "drawing board" codes often do need refinement to really work (blablabla ... you get it, it's likely full of bugs LOL). So, it would be great if QGC could be extended to allow for that to happen. 

What do you say?

dogmaphobic, 2019-05-16:
It's a bit of a chicken-and-the-egg issue. I have nothing to test with. In theory it's simple. If I see `mavlinkftp:///definition.xml` in the URL, I would use MAVLink FTP, which in itself is straight forward. With that said, it will be a few weeks before I have the time to deal with this.

DonLakeFlyer, 2019-05-16:
Keep in mind that the current MAVLink FTP implementation in QGC is crap.

dogmaphobic, 2019-05-16:
Also note that it's been temporarily removed (it was QtWidgets)

MatejFranceskin, 2019-05-16:
I am currently testing and using Mavlink FTP between QGC and the pairing manager with the idea to later include it also in the MavSDK and then in camera manager to transfer camera.xml file. In pairing manager I use the Mavlink FTP code from Firmware.

olliw42, 2019-05-16:
you guys made me laugh now ... it can't be that terrible :)

yeah, the chicken and the eggs ... 
I would be happy to do the testing, and I would not mind if it would have to go iterative. That is, I would be happy to test what you implement following the theory, and then to test the next theory, and  so on, until convergence :) 
(I suspect it wouldn't need many iterations though)

> If I see mavlinkftp:///definition.xml in the URL, I would use MAVLink FTP

this was already useful info ... 

> With that said, it will be a few weeks before I have the time to deal with this.

> Also note that it's been temporarily removed (it was QtWidgets)

things are how they are, no pressure

> Keep in mind that the current MAVLink FTP implementation in QGC is crap.

good occasion to improve it ;)  no, just kidding, no pressure

anyways, please go ahead as it's convenient for you, I'll be here

:)

olliw42, 2019-05-16:
> with the idea to later include it also in the MavSDK and then in camera manager to transfer camera.xml file

I'm looking forward for this to happen :)

DonLakeFlyer, 2019-05-16:
If mavlink ftp is going to be a thing. I would start over using the Log Download code as a basis for a generic block download backend which can handle/fill in missing blocks. Which can then be used behind both Log Download and mavlink ftp. Current mavlink ftp just fails if it has missing blocks.

closed 2026-10-04 by github-actions[bot]