mavlink/qgroundcontrol #7454: How to synchronously control multi uavs in QGC

myname1234, 2019-05-17:
I successfully build the multi-uav-launcher in ubuntu 16.04，and I used the latest QGC to control these vehicles. The QGC can connect all vehicles, but I can't control all the UAVs synchronously.When I use the multi vehicles mode, the UAV in gazebo take off or move one by one rather than  in the same time. So how can I fix  this issue.

DonLakeFlyer, 2019-05-17:
Not sure what you mean by synchonously? QGC is going to send the commands to the vehicles one at a time not instantansouly to each. So there is always going to be some slight delay in when they get to the vehicle.

myname1234, 2019-05-18:
> Not sure what you mean by synchonously? QGC is going to send the commands to the vehicles one at a time not instantansouly to each. So there is always going to be some slight delay in when they get to the vehicle.

When i control the vehicles, they can‘t take off or move at the same time,is there any solutions to reduce  the delay among the vehicles?

DonLakeFlyer, 2019-05-18:
Not with QGC

closed 2019-05-18 by DonLakeFlyer