mavlink/qgroundcontrol #7460: Video source options (RTSP, UDP) are not showing in android build while all the options are visible in windows build

NejatRmz, 2019-05-20:
Hi,

i have followed the instructions [here](https://github.com/mavlink/qgroundcontrol/tree/master/src/VideoStreaming) and downloaded gstreamer-1.0-android-universal-1.14.4.tar.bz2 file and added to directory as mentioned for android.
When i build the code for windows it works fine and shows the all the options in video source select box and i can start video stream through RTSP but in android build non of the video stream options are visible in android devices. You can see the capture [here](https://imgur.com/Ji3IUjs).
Is this related with GStreamer?

What am i missing?

Williangalvani, 2019-05-22:
Hello!
I had similar issues and I'm working on a Docker to build it:
https://hub.docker.com/r/williangalvani/qgc-android-build

It can build either master or from your local filesystem. Please let me know if you chose to use it and run into any issues.

bismahA, 2023-08-02:
I am having the same issue
Can anyone please guide how to fix this
Pleasee!!

zdanek, 2023-08-02:
@bismahA are you also using docker build?

bismahA, 2023-08-02:
No i acutally cloned the repo using this
git clone --recursive -j8 https://github.com/mavlink/qgroundcontrol.git

> @bismahA are you also using docker build?

HTRamsey, 2024-06-02:
This would have been related to how/if gstreamer was linked, which is pretty much automated in now in CMake

closed 2024-06-02 by HTRamsey