mavlink/qgroundcontrol #7462: Camera trigger sequence number visible in flight screen values

sanderux, 2019-05-20:
For peace of mind some users have requested to display 
CAMERA_TRIGGER.seq in the flight screen values.

Just to make sure the vehicle is actually sending the triggers, it would be a shame to fly almost 2 hours with nothing to show for it.

https://mavlink.io/en/messages/common.html#CAMERA_TRIGGER 

/cc @DonLakeFlyer

DonLakeFlyer, 2019-06-04:
QGC already shows you camera triggers with an icon on the fly view. Does adding a number to that really add value? It's going to add a lot of clutter. You should see new ones show up as the vehicle flies.

sanderux, 2019-06-06:
Icon? what icon? i've never seen that.
In any case, if you would add a section "CAMERA" on the values screen people can simply choose what to display. If mapping is all they do then that info would be highly relevant and very little more would be shown.

DonLakeFlyer, 2019-06-06:
QGC already uses the CAMERA_IMAGE_CAPTURED(px4) or CAMERA_FEEDBACK(ArduPilot) telemetry from the vehicle to display camera trigger icons on the map at the specified lat/lon position. And the at the end of a mission it will display the number of images captured.

[image]

DonLakeFlyer, 2019-06-15:
I"m closing since QGC shows camera trigger icons already.

closed 2019-06-15 by DonLakeFlyer