mavlink/qgroundcontrol #7464: Select language invalid on Daily build

bys1123, 2019-05-21:
## Expected Behavior
UI shows right language after language selection.

## Current Behavior
Nothing changed after language selection.

## System Information
When posting bug reports, include the following information
- Operating System: [Android, Windows]
- QGC Version: [master]
- QGC build: [daily]

## Detailed Description
I saw language support have lots of change these days, but I think language files are still on master, but is not valid.

## Log Files and Screenshots
[image]

dogmaphobic, 2019-05-21:
It must be a platform issue. It is working on macOS but I have not had a chance to try elsewhere.

[image]

tops4u, 2019-05-21:
Same here on MacOS. Switched to German, but nothing happened - still english.

hamishwillee, 2019-05-21:
When was this build generated? If it is today, then only approved translations are being exported from QGC, so on some languages that would be "none".

closed 2024-06-02 by HTRamsey