mavlink/qgroundcontrol #7466: Crash when changing Command in Plan Mode

tops4u, 2019-05-21:
Happening on Stable and Current Master.

## Expected Behavior
When selecting "Change Command" I can see a panel sliding in from the right but crashes before fully expanded:

[image]

## Steps to Reproduce:
Plan a simple mission with at least 1 Waypoint
Click on "Change Command" 
QGC Crashes

## System Information
- MacOS 10.14.4
- QGC Version: Current Stable and Current Master (as of 21.5.2019)

## Log Files and Screenshots
System Error Report:
```
Process:               QGroundControl [3351]
Path:                  /Users/USER/Desktop/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:            org.qgroundcontrol.qgroundcontrol
Version:               LastQtWidgetBuild.135.g4d55d0c52 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           QGroundControl [3351]
User ID:               501

Date/Time:             2019-05-21 11:33:57.520 +0200
OS Version:            Mac OS X 10.14.4 (18E226)
Report Version:        12
Anonymous UUID:        98924060-6243-6329-0C77-72B13289956F

Sleep/Wake UUID:       B56E3E90-A015-494B-8DDD-A4FD7328C08D

Time Awake Since Boot: 17000 seconds
Time Since Wake:       1200 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000070
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [3351]

VM Regions Near 0x70:
--> 
    __TEXT                 000000010a3b1000-000000010b999000 [ 21.9M] r-x/rwx SM=COW  /Users/USER/Desktop/QGroundControl.app/Contents/MacOS/QGroundControl

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.qgroundcontrol.qgroundcontrol	0x000000010a694363 0x10a3b1000 + 3027811
1   org.qgroundcontrol.qgroundcontrol	0x000000010a694afc 0x10a3b1000 + 3029756
2   org.qt-project.QtQml          	0x000000010e281f42 0x10e12c000 + 1400642
3   org.qt-project.QtQml          	0x000000010e27e7dc 0x10e12c000 + 1386460
4   org.qt-project.QtQml          	0x000000010e27df4e QV4::QObjectMethod::callInternal(QV4::Value const*, QV4::Value const*, int) const + 1342
5   org.qt-project.QtQml          	0x000000010e293cfa QV4::Runtime::method_callProperty(QV4::ExecutionEngine*, QV4::Value*, int, QV4::Value*, int) + 906
6   org.qt-project.QtQml          	0x000000010e28a5dd 0x10e12c000 + 1435101
7   org.qt-project.QtQml          	0x000000010e2937a0 QV4::Runtime::method_callName(QV4::ExecutionEngine*, int, QV4::Value*, int) + 224
8   org.qt-project.QtQml          	0x000000010e28a7ca 0x10e12c000 + 1435594
9   org.qt-project.QtQml          	0x000000010e33d892 QQmlJavaScriptExpression::evaluate(QV4::CallData*, bool*) + 594
10  org.qt-project.QtQml          	0x000000010e2d4fd8 QQmlBoundSignalExpression::evaluate(void**) + 1224
11  org.qt-project.QtQml          	0x000000010e2d57dd 0x10e12c000 + 1742813
12  org.qt-project.QtQml          	0x000000010e320105 QQmlNotifier::emitNotify(QQmlNotifierEndpoint*, void**) + 757
13  org.qt-project.QtCore         	0x000000010d9c777d QMetaObject::activate(QObject*, int, int, void**) + 125
14  org.qt-project.QtQml          	0x000000010e35649d QQmlObjectCreator::finalize(QQmlInstantiationInterrupt&) + 1197
[... 657 lines cut ...]
__DATA                            47.4M      466 
__FONT_DATA                          4K        1 
__GLSLBUILTINS                    5176K        1 
__LINKEDIT                       245.3M       90 
__TEXT                           371.4M      456 
__UNICODE                          564K        1 
mapped file                      688.0M      284 
shared memory                      640K       13 
===========                     =======  ======= 
TOTAL                              2.1G     1962 
TOTAL, minus reserved VM space     1.9G     1962 

Model: MacBookPro14,1, BootROM 190.0.0.0.0, 2 processors, Intel Core i5, 2.3 GHz, 16 GB, SMC 2.43f6
Graphics: kHW_IntelIrisGraphics640Item, Intel Iris Plus Graphics 640, spdisplays_builtin
Memory Module: BANK 0/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
Memory Module: BANK 1/DIMM0, 8 GB, LPDDR3, 2133 MHz, 0x802C, 0x4D5435324C31473332443450472D30393320
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x170), Broadcom BCM43xx 1.0 (7.77.61.2 AirPortDriverBrcmNIC-1305.8)
Bluetooth: Version 6.0.11f4, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Network Service: Thunderbolt Bridge, Ethernet, bridge0
Network Service: VPN, PPP (L2TP), ppp0
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 39.2

```

DonLakeFlyer, 2019-06-11:
I can get master to crash but I can't get Stable to crash.

DonLakeFlyer, 2019-06-11:
From fixing ths crash this would not have affect Stable.

closed 2019-06-11 by DonLakeFlyer