mavlink/qgroundcontrol #7471: Camera Manager: Collection of behaviors to imporve

olliw42, 2019-05-23:
I've made much progress with implementing a CameraManager support into the STorM32 gimbal controller to the point that it is practically usable, and in the practical tests I start to observe various behaviors of the camera manager GUI implementation which - IMHO - could be improved for better user friendliness. Instead of raising an issue for each, I thought I rather collect them here (pl tell if you prefer single issues for each).

1) Stream Selector field shown even if no STREAM capability

See #7450.

2) In photo mode single shot or interval mode should be indicated.

It's easy to forget what the last setting for the photo mode (single or interval) was, and not remembering it can lead to unexpected behavior. I thus suggest to have a graphical element on the GUI's "frontpage" showing this. E..g. this often seen "multiple-clocks" graphics.

3) Don't allow changes of camera mode (photo vs video) while a video is recording or a a photo interval is running.

Currently that's possible, and results in strange states for the red button.

4) In photo mode provide a graphical feedback when the next photo can be taken.

In photo mode with single shots the response is very slow, i.e. it takes few seconds before the next photo can be taken. If a hit on the red button was successful currently can only be inferred indirectly through an increasing number of photos. I suggest to provide some graphical mechanism to show to the user when it is ready for the next shot. E.g., the red button could get square, and round again when ready for the next shot.

5) Clarify the behavior and relationship as regards CAM_MODE and CAMERA_CAP_FLAGS.

Depending on the capability flags and the existence of a CAM_MODE parameter various GUI configurations result. However, not all combinations lead to what I would consider the expected configuration and behavior of the GUI. E.g. from the description for the CAMERA_CAP_FLAGS_HAS_MODES flag I would have expected that with CAMERA_CAP_FLAGS_CAPTURE_VIDEO + CAMERA_CAP_FLAGS_CAPTURE_IMAGE but no CAMERA_CAP_FLAGS_HAS_MODES and no CAM_MODE parameter the GUI would still show video and photo modes, which is not the case.

Thx for all your work on this, the CameraManager is really super cool.

MatejFranceskin, 2019-05-23:
There are plans to improve and add features to camera manager in the near future and your suggestions will be considered. We could use some help in testing it afterwards.
Could you share your configuration xml file so that we see how you're using it?

olliw42, 2019-05-23:
the STorM32 controller provides a multi-purpose camera support. So, some supported cameras don't have a xml nor a CAM_MODE. Most of them are actually of this type (I've named them 'STorM32 Generic Camera'). This are actually most of them, and for them my implementation is working well now (I believe). I do have a couple of test camera implementations, which use xmls without and with CAM_MODE, and also different capability and storage flags.
The two xmls with which I'm currently playing are these: www.olliw.eu/drop/storm32mavcamera_dummy1.xml and www.olliw.eu/drop/storm32mavcamera_runcamsplitmini2.xml. But note, these may change as I progress.

MatejFranceskin, 2019-05-23:
I will contact you when I make some changes just to make sure that I don't break anything in your setup.

olliw42, 2019-05-23:
this would be most kind, but I would be tracking them most likely myself too.

btw, I'm currently using QGCs daily from 2019-05-15. I tried a later daily but the QtQuick conversion appears to not yet be fully completed. So, I'm watching this anyhow, and would go on with a newer QGC iff it appears to have converged, at which time also your changes might start to come in. This issue here I have not raised with any expectation but to not lose the observations. 

Don't worry with breaking anything in my setup. If that should happen I will happily adapt my codes. The end result is what counts, namely a super cool camera manager GUI :)

olliw42, 2019-06-01:
just for info, this is what I'm talking about when I'm saying "I've made much progress with implementing a CameraManager support into the STorM32 gimbal controller to the point that it is practically usable":

https://youtu.be/nGszTJCzSMk

I'm looking forward to help with testing any improvements in the Camera Manager part :)

closed 2026-10-04 by github-actions[bot]