mavlink/qgroundcontrol #7474: Corrupted log file

kaklik, 2019-05-25:
I want to replay [attached log file](https://github.com/mavlink/qgroundcontrol/files/3219441/2019-05-22.18-20-20.zip) to view exact command sequence from QGC.   

## Expected Behavior
Open the log file and examine the situation by time slider. 

## Current Behavior
After click on Replay Flight Data button and select the log file. The following message  pops

```
Log Replay Error
The log file '2019-05-22 18-20-20.tlog' is corrupt. No valid timestamps were found at the end of the file.
```

[image]

## Steps to Reproduce:

1. Proceed flight with PX4 UAV
2. Turn off UAV
3. Disconnect telemetry modem from USB
4. Close ground control's station laptop
5. Tranport everything to the workshop
6. Open laptop again and try to replay logged data. 

## System Information

- Operating System:  Ubuntu 18.04.2 LTS
- QGC build: Development HEAD:49ef8c141 2019-05-15
- Flight Controller: Pixhawk 3
- Autopilot (with version): PX4 v1.9.0 (RC)

bigbellmercy, 2019-05-28:
This issue is important because when a drone is crashed far away and this bug happens we cannot find the drone since we don’t know its location. As a case, it happened to me.

DonLakeFlyer, 2019-06-05:
@bigbellmercy Can you try the AppImage from here: https://qgroundcontrol.s3-us-west-2.amazonaws.com/DonLakeFlyer/QGroundControl.AppImage?

DonLakeFlyer, 2019-06-05:
This should be fixed now. New daily builds will be available in a few hours. Stable will be updated as well but that won't happen for a bit.

DonLakeFlyer, 2019-06-05:
@kaklik Flagged the wrong person. Can you try the new build posted above?

closed 2019-06-05 by DonLakeFlyer

kaklik, 2019-06-05:
@DonLakeFlyer  Yes, I could confirm that build from above is able to open the corrupted log file. Thanks for fixing the bug. :)

DonLakeFlyer, 2019-06-05:
Great, thanks for verifying