mavlink/qgroundcontrol #7478: Crashed when start prompting "Cannot create a QWidget without QApplication"

wuxianshen, 2019-05-29:
I built the QGC from source without any changes in QT 5.11.3, the compilation is finished successfully but when I try to start QGC with the Run button of QTCreator, some errors come.
Application Output prompts that :
QWidget: Cannot create a QWidget without QApplication.

Has anyone ever met this problem before? Is there any problem with my QT installation?
Thanks!

## System Information
When posting bug reports, include the following information
- Operating System: [Ubuntu 16.04]
- QGC Version: [master for 2018.5.28, v3.5.2]
- QGC build: [self-built from source]
- Qt Version: [5.11.3]

dogmaphobic, 2019-05-29:
Yes, this is known issue. @stefandunca did you merge those fixes to QGC master?

stefandunca, 2019-05-29:
@dogmaphobic here it is: https://github.com/mavlink/qgroundcontrol/pull/7479

closed 2019-07-16 by wuxianshen