mavlink/qgroundcontrol #7486: Hover and capture

Stifael, 2019-06-03:
## feature request
It would be great if hover-and-capture depends on the exposure time of the camera.

dogmaphobic, 2019-06-03:
There is no way for QGC to be aware of exposure time though. In addition, is any gimbal good enough to maintain a camera steady for a long exposure? How long of an exposure are we talking about?

Stifael, 2019-06-03:
> There is no way for QGC to be aware of exposure time though

It can just be part of the settings for each camera. 

>  In addition, is any gimbal good enough to maintain a camera steady for a long exposure?

It is not gimbal dependent, but rather depends on the hover time of the vehicle. For instance, at night you want to increase the shutter speed and hence the hover time should be increased as well.

dogmaphobic, 2019-06-03:
>It can just be part of the settings for each camera.

QGC itself has no knowledge of exposure. It's a (custom) camera parameter and not something that can be set from a mission command. It's not impossible to implement but I wonder how useful this is given the effort it would take.

> It is not gimbal dependent, but rather depends on the hover time of the vehicle. For instance, at night you want to increase the shutter speed and hence the hover time should be increased as well.

It is very gimbal dependent if you are talking about exposures greater than 100ms. The camera has to be absolutely still for that long. I can't imagine a situation where very long exposures (greater than 1 second) would ever work.

Stifael, 2019-06-03:
> It is very gimbal dependent if you are talking about exposures greater than 100ms.

Not in the case that I describe. For this issue, you can assume the gimbal is perfect.
The issue is really that you absolutely don't want to move (forward flight) during capture. This is particularly important during the night, where the exposure time needs to be increased. 

> It's not impossible to implement but I wonder how useful this is given the effort it would take.

If I set a survey with hover-and-capture and trigger distance 2m and there are no additional settings, then I expect the drone to hover-and-capture long enough without creating blurry images. We could also add a setting where the user has to set the hover time himself, or the survey is generated automatically with hover time based on the already provided settings.

dogmaphobic, 2019-06-03:
> If I set a survey with hover-and-capture and trigger distance 2m and there are no additional settings, then I expect the drone to hover-and-capture long enough without creating blurry images.

Alright, this makes sense and it is not related to "exposure" time. That is, what you are requesting is that when you tell the mission to _hover_ and capture, truly stop the drone before triggering the camera and resuming flight. The duration of the hover period won't ever be over a second given that it's unreasonable to think a drone can take a very long exposure without movement (and it will take longer than that to fully stop and become static.) @DonLakeFlyer will need to chime in here. The flight controller would be the one to know when the drone is static and only then issue the camera trigger command.

Stifael, 2019-06-03:
> That is, what you are requesting is that when you tell the mission to hover and capture, truly stop the drone before triggering the camera and resuming flight

Partially. There are two issues:
1.) ensure that drone is fully stopped (the autopilot can handle that)
2.) take a picture and hover `long enough` before continuing (https://github.com/mavlink/qgroundcontrol/issues/7488). Here there needs to be some metrics to decide what `long enough` means. 

In one of the tests that we conducted at night, we changed the shutter speed of the camera. The blurry of the images was aligned with the motion of the vehicle, indicating that the drone was already in forwarding flight during capture.

DonLakeFlyer, 2019-06-04:
This requires two (or maybe more) new mavlink commands:
* A waypoint the vehicle flies exactly to, no acceptance radius and does not allow to mission state machine to proceed until vehicle is stable
* A camera trigger which holds the mission state machine until the image taking is fully complete

Stifael, 2019-06-13:
> This requires two (or maybe more) new mavlink commands
> * A waypoint the vehicle flies exactly to, no acceptance radius and does not allow to mission state machine to proceed until vehicle is stable

Can that not be solved with the current message? Is there ever the case where someone wants to do hover-and-capture (not fly-through) and does not care about the precise location?

closed 2026-10-04 by github-actions[bot]