mavlink/qgroundcontrol #7491: Android app: add tracks.

AndKe, 2019-06-04:
The flight path is not visible in Android app, flying over a feature-less area, like from a ship/over sea, makes it impossible have an understanding of past search-paths.

DonLakeFlyer, 2019-06-04:
Wierd. Do you know is this started happening recently. Not much change in that code.

AndKe, 2019-06-04:
Sorry, I do not know. It is unusual that I use QGC for those bigger things, especially at sea.
I have made an MAVProxy module that updates home point along with the ship, and discovered that AP2 did not display updated home location.
Just for fun, I tested QGC, which updated home just fine, and made a track, then connected my phone to the GCS computer over wifi, and discovered that there were no visible tracks on the screen.

DonLakeFlyer, 2019-06-05:
I tried both daily and stable builds doing this:
* Takeoff
* GoTo Location

In both cases I got vehicle tracks.

AndKe, 2019-06-05:
I think I know what happened.
Like I mentioned, this was a test, we did not flew, the ship(and UAV) was moving... It is likely that the tracks are only drawers while flying... Correct ?

DonLakeFlyer, 2019-06-05:
Only when armed. Which I think still makes sense. So things are ok here.

closed 2019-06-05 by DonLakeFlyer