mavlink/qgroundcontrol #7494: "Parameters are missing from firmware" but doesn't identify which

lecw, 2019-06-05:
Current : As in the screenshot error message, sometimes QGC complains about missing parameters, but doesn’t tell me which parameters are missing (other than “-1”). 
Expected : if some parameters are actually missing (which I'm not sure of), the error message should say which.

## Steps to Reproduce:
1. plug in the pixhawk, wait for it to boot
2. some messages appear (HIL is enabled, etc.), then this one

## System Information
- Ubuntu Mate 16 or 18.04
- QGC Version: the latest one as of 2019-05-17
- QGC build : daily I think, else stable
- Flight Controller: Pixhawk 4
- Autopilot (with version): PX4 1.8.2 with a few custom modifications.

The same firmware will _not_ trigger this warning on some other versions of QGC on different computers (such as on windows, or in the stable-vs-daily one).

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
[image]

DonLakeFlyer, 2019-06-05:
Do you have console logs? The link above goes to the docs? If not, can you attach them?

lecw, 2019-06-08:
No I didn't have them, and now I can't reproduce it. Now I run into "Firmware is not latest stable version ! Firmware version is -1.-1.-1, latest stable is 1.9.0.".

[image]

We have two computers (both Ubuntu) that both pull our firmware from the same github and push it to the same pixhawk 4 (with `make px4fmu-v5_default upload`). In one case, the firmware is then recognized by QGC as 1.8.2.dev, but in the other case, firmware is "Unknown" and yields this "-1.-1.-1 version" warning. That seems related to me, tell me if not.

A console log for that is here :
[log-1-1-1.txt](https://github.com/mavlink/qgroundcontrol/files/3267991/log-1-1-1.txt)

DonLakeFlyer, 2019-06-08:
> but in the other case, firmware is "Unknown" and yields this "-1.-1.-1 version" warning. That seems related to me, tell me if not.

Your firmware is not reporting correct version information in the AUTOPILOT_VERSION message.

DonLakeFlyer, 2019-06-11:
Without a repro I'm closing this.

closed 2019-06-11 by DonLakeFlyer