mavlink/qgroundcontrol #7503: Proposal for Improving User-friendliness of Component Handling, GCS Side

olliw42, 2019-06-06:
In this proposal to MAVLink, [Proposal for Improving User-fiendliness of Component Handling](https://github.com/mavlink/mavlink/issues/1133), I proposed to establish a mechanism similar to that used by the Camera Manager microservice also for components in general. This would allow a GCS to retrieve relevant meta data by fetching a component definition xml file, which it can use to provide the user with a nicer and much more user-friendly interface. 

A typical application example, which was the initiating idea, would be to enrich the presentation of the component parameters in a similar way as it is done for the PX4 autopilot, and reach a similar level of convenience. Currently a user only sees the name-value pair, which isn't 2019, as I would express it.

The proposal was positively received by MAVlink, and in a dev call it was decided to progress iteratively, and start with tentatively defining some MAVlink message (see below), and then see how it goes.

However, obviously, defining a MAVlink message is the easy part, the harder part is to establish such a functionality also in a GCS, and that's why I am here. :) :) :)

I have recently added a MAVlink Camera microservice to the STorM32 gimbal controller, which exploits QGC's Camera Manager (e.g. https://youtu.be/nGszTJCzSMk). So, this component has the relevant pieces available and working, and I'm currently adding support of the said MAVLink message for components to it. Specifically, I'm currently sitting over creating the component definition xml file.

I've analyzed the methods for describing parameter meta data used by QGC/PX4, QGC/APM, MissionPlanner, and STorM32, and found that they in fact use very similar meta data - a biologist would call that convergent evolution I guess :). However, the meta data is stored in very different formats/language constructs. The highly similar meta data strongly suggests that a universal format should be developed and be used by these entities, and also be used in the component definition file. However, this would add another level of challenge, and I thus think one should just start with using QGC's own PX4 xml format. This should make it much easier for you to get a working prototype, since one "just" has to take over what is there for the PX4 autopilot, and I believe that changing the xml anytime later will not be very difficult. 

So, I'm sitting over creating a STorM32 definition xml file in QGC/PX4 format.  And it's here there I would have some first questions, but I will put them in a next post.

I conclude this post with the MAVLink command definition I tentatively designed:

```
<enums>
  :
    <enum name="MAV_CMD">
    :
    <entry value="700" name="MAV_CMD_REQUEST_COMPONENT_INFORMATION" hasLocation="false" isDestination="false">
    <description>Request component information (COMPONENT_INFORMATION).</description>
        <param index="1" label="Capabilities" minValue="0" maxValue="1" increment="1">0: No action 1: Request component information</param>
        <param index="2">Reserved (all remaining params)</param>
    </entry>
    :
  </enum>
  :
</enums>
<messages>
    :
    <message id="700" name="COMPONENT_INFORMATION">
        <description>Information about a component</description>
        <field type="uint32_t" name="time_boot_ms" units="ms">Timestamp (time since system boot).</field>
        <field type="uint8_t[32]" name="vendor_name">Name of the component vendor</field>
        <field type="uint8_t[32]" name="model_name">Name of the component model</field>
        <field type="uint32_t" name="firmware_version">Version of the component firmware (v &lt;&lt; 24 &amp; 0xff = Dev, v &lt;&lt; 16 &amp; 0xff = Patch, v &lt;&lt; 8 &amp; 0xff = Minor, v &amp; 0xff = Major)</field>
        <field type="uint32_t" name="hardware_version">Version of the component hardware (v &lt;&lt; 24 &amp; 0xff = Dev, v &lt;&lt; 16 &amp; 0xff = Patch, v &lt;&lt; 8 &amp; 0xff = Minor, v &amp; 0xff = Major)</field>
        <field type="uint32_t" name="capability_flags" enum="COMPONENT_CAP_FLAGS">Bitmap of component capability flags.</field>
        <field type="uint16_t" name="component_definition_version">Component definition version (iteration)</field>
        <field type="char[140]" name="component_definition_uri">Component definition URI (if any, otherwise only basic functions will be available).</field>
    </message>
    :  
</messages>
```

olliw42, 2019-06-06:
few questions as regards the QGC/PX4 xml details (I'm looking at the file PX4ParameterFactMetaData.xml).

I find that it only uses `type="FLOAT"` and `type="INT32".` I also find that there can be a `<decimal>` and <increment> entry for FLOAT-type parameters, but not for INT32-type. 

That it doesn't use e.g. `type="UINT8"` but only `"INT32"`,  does this mean that PX4 indeed has only integer-valued parameters of type `MAV_PARAM_TYPE_INT32`, or does it mean that QGC of course takes care of a `MAV_PARAM_TYPE_UINT8` when writing/reading parameters to/from the component, but treats them all alike only for the purpose of displaying/handling them in the GUI?

If it's the latter case, does this expand across float - integer boundary, or is it distinct for float and integer? This is maybe better clarified by a specific example (which exists in the case of the STorM32, and which I think is realistic and relevant even beyond the STorM32) :

Let's consider an integer-valued parameter `MAV_PARAM_TYPE_UINT8`, which can range from 0 ... 100, but should be displayed as 0.0 ... 10.0.

Within the available xml constructs, I could see two ways. One is to use `type="FLOAT"` and `<decimal>1</decimal>`, knowing that QGC would convert it to/from `MAV_PARAM_TYPE_UINT8` for storing/reading. The second one that the field `<decimal>` can also be used for INT32-type parameters. 

Which one would be the way to go?  Is this currently possible at all?

Let's now consider a slightly more complicated case. Let's consider an integer-valued parameter `MAV_PARAM_TYPE_UINT8`, which can range from 0 ... 100, should be displayed as 0.00 ... 1.00, and should be incremented in steps of 0.05. The question would be as before. Would it work with FLOAT and decimal=2 and increment=0.05, or would it work with INT32 and decimal=2 and increment=0.05, or doesn't it currently work at all.

If none of this doesn't currently work, I would strongly suggest to extend the mechanism to support this. I believe that this is a very realistic and situation, and will come up later or sooner anyhow.

DonLakeFlyer, 2019-06-06:
QGC supports all possible MAV_PARAM_TYPE values as specified by the parameter protocol both internally throughout the code and in the PX4 Parameter Meta Data parser.

> Let's consider an integer-valued parameter `MAV_PARAM_TYPE_UINT8`, which can range from 0 ... 100, but should be displayed as 0.0 ... 10.0.

This is not currently possible. You can't do that sort of translation back/forth.

DonLakeFlyer, 2019-06-06:
> I specifically suggest to add an attribute 'dispname' (or similar), which if present is used by QGC for displaying as parameter name, instead of 'name'.

Why is that needed in addition to <short_desc> which is what is also show in the parameter editor list?

olliw42, 2019-06-06:
> QGC supports all possible MAV_PARAM_TYPE values as specified by the parameter protocol both internally throughout the code and in the PX4 Parameter Meta Data parser.

so, if I understand you correctly, the type in the PX4 `<parameter>` field must be identical to the MAV_PARAM_TYPE of the parameter, i.e. for the above example one has to write type="UINT8". Correct?

>This is not currently possible. You can't do that sort of translation back/forth.

sad. Let's hope that possibility is going to be added eventually :)

> Why is that needed in addition to <short_desc> which is what is also show in the parameter editor list?

I'm not sure I understand the source of confusion here. Let me try to describe more. The user would never see the cryptic mavlink parameter name but only the dispname. 

For e.g. the main param list, the user is currently, that's at least my understanding, presented with 

`mavlink-parameter-name   |  value | short-description`

Now the user would be presented with

`dispname | value | short-description`

So, the intended task can't be accommodated with short_desc. E.g. instead of 

`STP_YAW_MOTUSG  |  3 |   Selects the operation mode of the yaw motor.`

she would see 

`Yaw Motor Usage  |  3 |   Selects the operation mode of the yaw motor.`

and in fact ideally she would see 

`Yaw Motor Usage    |     Enabled  |  Selects the operation mode of the yaw motor.`

In my opinion the last is by miles more user-friendly. And moreover, if in doubt the user could look into any document of the component and find a description for Yaw Motor Usage, or may know what it is anyhow from having experience with the components native GUI or tools. Chances are that even google my help. All this is much less likely so for the cryptic mavlink name. Again, in my opinion by miles more user-friendly.

olliw42, 2019-06-06:
if I would add an attribute 'dispname' even when not supported, would this confuse/break QGC?

when I would use the fields `<increment>` and `<decimal>` for an integer-type parameter, would this confuse/break QGC?

olliw42, 2019-06-07:
I've done now "my" stuff, i.e., have support of the component definition file in STorM32, and also created xml files to start with
www.olliw.eu/drop/storm32mavcomponent_foc.xml
www.olliw.eu/drop/storm32mavcomponent_nonfoc.xml
(there are two of them since that's how the STorM32 works, but this shouldn't bother you, the request returns the correct uri for the current STorM32 mode)
I couldn't make the STorM32 firmware available yet for time, and am going for vacation soon for few days. But I think the MAVLink message speaks for itself and the xml's should give sufficient idea.

would be cool to see this going forward :)

DonLakeFlyer, 2019-06-07:
> if I would add an attribute 'dispname' even when not supported, would this confuse/break QGC?

QGC ignores tags it doesn't know about.

> when I would use the fields `<increment>` and `<decimal>` for an integer-type parameter, would this confuse/break QGC?

<increment> is support on any value. That said it only comes into play if the value is repsented int he ui as a slider which is only done on a couple tuning parameters. 

<decimal> on an int would be parsed and then set in the meta data but ultimately do nothing with them. That said, it could leads to possible bugs if QGC code is incorrect  or changes. In reality the meta data parse should fail if it hits this on an int value.

olliw42, 2019-06-07:
in the above .xml's you'll find `<increment>` included, since
- I really think it should be trivial to adapt the QGC code such that they are ignored then QGC doesn't want to use them
- gives you guys the option to add such support
This would have been true for `<decimal>` too, but I wasn't 100% sure what they do, so left them out.

>That said it only comes into play if the value is repsented int he ui as a slider which is only done on a couple tuning parameters.

in my QT STorM32 App (which is outdated now however) one can both enter the value and set it through sliders, for all params (except the drop list ones). I found that very convenient.

DonLakeFlyer, 2019-06-07:
> * for `<decimal>` too, but I wasn't 100% sure what they do

<decimal> specifies how many decimal places you want to display on a real number:
<decimal> 1 -> 10.1
<decimal> 2 -> 10.12

olliw42, 2019-06-07:
many thx
this was my guess but I couldn't confirm, thx for clarifying
I could quickly update the xml's to include them too, if that is desirable