mavlink/qgroundcontrol #7514: undefined keyword or key name: LastQtWidgetBuild

alimollaye, 2019-06-11:
hello. i'm new in qt and i try to compile QGC in win10 . everything is ok except an Error that i can't figure it out. this is error Message:
E:\QGroundContol_project\Working_Version\build-qgroundcontrol-Desktop_Qt_5_11_0_MSVC2015_32bit-Release\QGroundControl_resource.rc:6: error: RC2104 : undefined keyword or key name: LastQtWidgetBuild
i skipped the line in resourece file that were related to "LastQtWidgetBuild" and everything went well but at the end of compile again another error is showed . error message is:
 LNK1117: syntax error in option 'VERSION:LastQtWidgetBuild.147'

what should i do? 
anyone can help me please.

DonLakeFlyer, 2019-06-15:
I believe this has been fixed in latest master

closed 2019-06-15 by DonLakeFlyer

DonLakeFlyer, 2019-06-15:
It's cause by the git tag LastQtWidgetBuild. If you remove the tag from your local repo it should build. The tag has been removed from master.