mavlink/qgroundcontrol #7516: MacOs: joystick crash

RyanHurst, 2019-06-12:
## Current Behavior
QGC crashes when using a Playstation DualShock 4(DS4) controller

## Steps to Reproduce:
1. Open QGC
2. Plug in DS4 controller
3. Connect to vehicle
4. Calibrate controller joysticks
5. Fly a drone around until QGC crashes

## System Information
- Operating System: Mojave 10.14.
- QGC build: Daily 683ffe673 and 3.5.2 release
- Flight Controller: pix racer
- Autopilot (with version): PX4 1.9.0

## Detailed Description
QGC keeps crashing while trying to fly a drone with a DS4 controller plugged in

## Log Files and Screenshots
[crash 1.txt](https://github.com/mavlink/qgroundcontrol/files/3282251/crash.1.txt)
[crash 2.txt](https://github.com/mavlink/qgroundcontrol/files/3282252/crash.2.txt)
[crash 3.txt](https://github.com/mavlink/qgroundcontrol/files/3282253/crash.3.txt)
[crash 4.txt](https://github.com/mavlink/qgroundcontrol/files/3282254/crash.4.txt)
[crash 5.txt](https://github.com/mavlink/qgroundcontrol/files/3282255/crash.5.txt)
[crash 6.txt](https://github.com/mavlink/qgroundcontrol/files/3282256/crash.6.txt)

DonLakeFlyer, 2019-06-15:
Hmm, there isn't really anything on the QGC side going on here. It's all in the library we use for joystick support and/or OSX. Which unfortunately means there is much we can do. Let me look if there is a newer SDL lib which can be used.

DonLakeFlyer, 2019-12-24:
Closing in the hope new SDL update fixed this

closed 2019-12-24 by DonLakeFlyer