mavlink/qgroundcontrol #7517: Strange audio language on Ubuntu 18.04 + workaround

ArkadiuszNiemiec, 2019-06-12:
## Expected Behavior
I expect the QTextToSpeech to use system locale to generate audio or TTS.

## Current Behavior
I get English TTS in a strange accent which is almost not understandable.

## Steps to Reproduce:
1. Fresh Ubuntu 18.04 configured with US locales (installed in Poland and connected to local wifi)
2. Install QGroundControl (daily or stable)
3. Connect to PX4
4. Listen

## System Information
- Ubuntu 18.04
- QGC build: Daily, stable or even builded from source

## Detailed Description
I have checked the `speech-dispatcher` and it works with `espeak`. I can confirm that it works perfectly by typic `spd-say "Hold flight mode"` All builds of QGroundControl have strange English accent and I cannot find a source of it.

In the meantime I have found a workaround: I can get clean English by adding this line:
```
_tts->setLocale(QLocale())
```
to https://github.com/mavlink/qgroundcontrol/blob/eee967c549d4958907f97c716f98b1c44c9b963f/src/Audio/AudioOutput.cc#L19-L24

Even stranger: I have checked the `_tts` in the debugger and it shows `en_US` before `_tts->setLocale(QLocale())` but still it fixes it.

I can also confirm that I have `en_US` locale by watching QGroundControl logs:
```
System reported locale: QLocale(English, Lating, UnitedStates) "en_US"
```

ArkadiuszNiemiec, 2019-06-12:
This bug report confirms my theory: https://bugreports.qt.io/browse/QTBUG-73552

dogmaphobic, 2019-06-13:
This is a dupe of some other issue but your bug research made a lot simpler to fix it. On it...

dogmaphobic, 2019-06-13:
Can you please try this and see if this makes any sense? As I explained in the PR, we may need to do the opposite (force it to use English always, regardless of your locale): https://github.com/mavlink/qgroundcontrol/pull/7518

dogmaphobic, 2019-06-13:
See https://github.com/mavlink/qgroundcontrol/pull/7518#issuecomment-501668327

ArkadiuszNiemiec, 2019-06-13:
Fixed by #7518

Niel66, 2019-08-23:
It was not fixed in the .appimage file
Here is the accent in which it spoke
[simplescreenrecorder-2019-08-23_17.36.26.zip](https://github.com/mavlink/qgroundcontrol/files/3534328/simplescreenrecorder-2019-08-23_17.36.26.zip)

cjffly, 2019-11-20:
Still have the same issue with the .appimage file and daily build

dogmaphobic, 2019-12-26:
I can't reproduce this on Ubuntu 18.04. I installed it using Italian:
```
System reported locale: QLocale(Italian, Latin, Italy) "it_IT"
Loading localization for "it_IT"
```
Text to Speech seems fine. That is, it is as crapy as every other TTS on Linux but it is using standard English.

DonLakeFlyer, 2020-02-08:
@tcanabrava When you crank up a 16.04 for that other problem. Could you try to reproduce this as well?

DonLakeFlyer, 2020-02-08:
I'm wondering if this is a 16.04 only problem and 18 is fine