mavlink/qgroundcontrol #7523: Cannot update firmware on new Pixhawk Cubes

eitan-babcock, 2019-06-13:
## Expected Behavior
Installing the firmware according to the standard installation steps should work.

## Current Behavior
While following the normal firmware upgrade steps, the Cube kicks out of the boot loader before the firmware can be loaded. It says "detected pixhawk Cube" and then immediately says "Upgrade cancelled" and goes to the Summary page and loads the parameters like a normal boot.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC
2. Click the Firmware tab
3. Plug in a Pixhawk Cube from ProfiCNC/Hex (a new one, older ones work fine)
4. You will not be able to upload the firmware, as it will kick out to the normal boot shortly after plugging in.

## System Information
When posting bug reports, include the following information
- Operating System: Ubuntu 16.04
- QGC Version: v3.5.2
- QGC build: Stable
- Flight Controller: Pixhawk 2.1 Cube
- Autopilot (with version): PX4 custom fork of 1.7, and also tried latest 1.9 stable release

## Detailed Description
Original discussion here: https://discuss.cubepilot.org/t/bootloader-issues-with-qgroundcontrol/1058

I’m running PX4 and QGroundControl, and on my most recent order of Cubes (Blue Cube, if that matters) I cannot upload the firmware from QGroundControl. I am able to do it from the command line, but when I try to upload through QGroundControl it kicks out of the bootloader before I have a chance to load the firmware.

I also noticed that the cubes take longer to boot than previous versions that didn’t have this issue.

When I run dmesg -w and plug in the device, it looks like my computer first sees it as a CubeBlack made by Hex/ProfiCNC, and then disconnects and reconnects viewing it as a PX4 FMU v2.x made by 3D Rogbotics.

```
[664522.631018] usb 1-1.4: new full-speed USB device number 49 using xhci_hcd
[664522.733000] usb 1-1.4: New USB device found, idVendor=2dae, idProduct=1001
[664522.733007] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[664522.733012] usb 1-1.4: Product: CubeBlack-BL
[664522.733016] usb 1-1.4: Manufacturer: Hex/ProfiCNC
[664522.733021] usb 1-1.4: SerialNumber: 3B0023001251383037393430
[664522.734339] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[664527.527407] usb 1-1.4: USB disconnect, device number 49
[664529.030896] usb 1-1.4: new full-speed USB device number 50 using xhci_hcd
[664529.133822] usb 1-1.4: New USB device found, idVendor=26ac, idProduct=0011
[664529.133831] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[664529.133836] usb 1-1.4: Product: PX4 FMU v2.x
[664529.133841] usb 1-1.4: Manufacturer: 3D Robotics
[664529.133846] usb 1-1.4: SerialNumber: 0
[664529.137767] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

eitan-babcock, 2019-06-13:
I also just tried the latest daily build of QGC. It had different behavior. It no longer immediately kicked out to a normal boot. I was able to go through the motions to upgrade firmware, but right when I clicked "OK" it seg faulted `Setmentation fault (core dumped)` and crashed the program.

DonLakeFlyer, 2019-06-13:
> but right when I clicked "OK" it seg faulted `Setmentation fault (core dumped)` and crashed the program.

That should be fixed if you download a new daily and install again.

eitan-babcock, 2019-06-13:
Confirmed, the newest daily build works in all regards for me.

Any idea when this will be pulled into a release? I need to know if I should have the rest of my development team pull the daily build or wait until the release.

DonLakeFlyer, 2019-06-13:
> Any idea when this will be pulled into a release?

Nope. At this point I don't know exactly what it is. Depends on how involved the difference is. It may not be possible/safe if too complex.

UnmannedAirman, 2019-06-25:
It appears this issue was resolved. Sorry, but what were the steps you guys took to resolve the issue? I have about 15 cubes that all have the issue described in the original post.

eitan-babcock, 2019-06-25:
The daily build from Jun 13 had a fix in place. All I did was install the daily build of QGC from that day and it worked. I imagine any daily builds after that day still work, but I haven't tried.
https://docs.qgroundcontrol.com/en/releases/daily_builds.html

mwiatt, 2019-10-18:
Experiencing the same issue here with QGC 3.5.5, Win10.

Old Cube, fine. New Cube, not detected during the firmware update process with 3.5.5. The daily build of QGC is working fine as well as the latest stable of Mission Planner. If I flash 1.9.2 using the daily QGC build and then go back 3.5.5, I get this error in the firmware update page after plugging in the 1.9.2 cube. The drivers are not being reinstalled when I am swapping back and forth between 3.5.5 and daily.

[image]

After the Cube is flashed with 1.9.2 using the daily QGC, it will communicate with 3.5.5 over USB, just not update firmware.

I noticed that the Cube is being ID's as "ProfiCNC CUBE F4 FMU" during bootloader phase:

[image]

and then "Legacy FMU" when PX4 is running:

[image]

The old Cube that works with 3.5.5 for programming IDs as a "Legacy FMU" during the bootloader phase as well as when PX4 is running.

Also, @DonLakeFlyer I saw in another issue that is dealing with Cube USB issues, that you thought all of the old drivers that start with "Windows Driver Package" were not being installed anymore. Not sure if this is related but, wanted to let you know that they are being installed:

[image]

DonLakeFlyer, 2019-10-18:
> that you thought all of the old drivers that start with "Windows Driver Package" were not being installed anymore.

That is most likely cause be bouncing back and forth between Stable and Daily installs.

DonLakeFlyer, 2019-10-18:
@mwiatt You are running into problems with the new ArduPilot bootloader which is on newer Cubes. Only daily builds have full support for that.

> The old Cube that works with 3.5.5 for programming IDs as a "Legacy FMU" during the bootloader phase as well as when PX4 is running.

This one doesn't have the ArduPilot bootloader on it. Hence it will work fine on Stable.

DonLakeFlyer, 2019-10-18:
Or in other words: This is all known behavior given the state of Stable and Daily.

mwiatt, 2019-10-18:
Thanks for confirming the known bootloader behavior.

FYI regarding drivers:

I uninstalled QGC, UAV Drivers, and 21x Windows Driver Package XXX. Then installed daily. I ended up with UAV Drivers + all 21x Windows Driver Package XXX.

I repeated the removal of everything and then installed 3.5.5. Same result with UAV Drivers + all 21x Windows Driver Package XXX.

In both cases, this pops up after the UAV Drivers installer starts but before it is finished:

[image]

Sorry to clutter up the thread as this is unrelated but I wanted to follow up since I brought it up.

DonLakeFlyer, 2019-10-18:
> I uninstalled QGC, UAV Drivers, and 21x Windows Driver Package XXX. Then installed daily. I ended up with UAV Drivers + all 21x Windows Driver Package XXX.

Hmm. That doesn't seem correct. Can you create a new issue with details and we can work from there.

SMenne, 2019-11-12:
I solved that Problem with Mission Planner / Load Customer Firmware / px4fmu-v3_default.px4 (which i before downloaded)

Dule-sat, 2020-11-27:
Hi everyone,

I just got my cube orange and have been experiencing the same issue. I am trying to flash px4 through QGC daily (tried on multiple devices and both on stabile and daily version).

I followed everything written here and on https://discuss.cubepilot.org/t/bootloader-issues-with-qgroundcontrol/1058/22 I am using Win 10 and downloaded daily release of qgc, however the issue is still present. 
Do you have some suggestions?
[image] 
Also you can see video here https://www.youtube.com/watch?v=S8k_tIdjeHk&feature=youtu.be

Thank you
Dusan

DonLakeFlyer, 2020-11-27:
Cube Orange bug is here: https://github.com/mavlink/qgroundcontrol/issues/9215. Provide a console log there to help with debugging.

Dule-sat, 2020-11-28:
> Cube Orange bug is here: #9215. Provide a console log there to help with debugging.

Can you please tell me how to provide console log? Thanks