mavlink/qgroundcontrol #7524: qgroundcontrol with gstreamer problem

focusonesoftware, 2019-06-13:
Hi,

I build groundcontrol w/o gstreamer on windows, It's OK.
I build qgroundcontrol with gstreamer on windows, I found "ASSERT failure in QSGNode::prependChildNode: "QSGGeometryNode is missing geometry", file scenegraph\coreapi\qsgnode.cpp, line 414 scenegraph\coreapi\qsgnode.cpp: 414".

windows 10
gstreamer version 1.14.4
qgroundcontrol master on 14/06/2019

Thank you.

DonLakeFlyer, 2019-06-15:
If you want to see how QGC creates daily builds from master look at https://github.com/mavlink/qgroundcontrol/blob/master/.appveyor.yml. This sequence of steps has not problem creating a build that works.

DonLakeFlyer, 2019-06-15:
Even though appveyor seems to be working. There is something strange going on here. Possibly only with debug builds. I can something similar to happen for me.

DonLakeFlyer, 2019-06-15:
Verified that release builds work. Only debug builds fail.

DonLakeFlyer, 2019-06-18:
Also Stable works fine. Also just realizing that appveyor has not been switched to 1.14.4. So it's using old version 1.5.2 installer. I've put up the 1.14.4 installers on to S3 for use. But I'll leave things as is for now. I'm switching Sentera stable to 1.14.4 though for testing.

jaxxzer, 2019-06-27:
Worth noting there is a more recent stable release of gstreamer: [1.16](https://gstreamer.freedesktop.org/releases/1.16/)

dogmaphobic, 2019-06-27:
> Worth noting there is a more recent stable release of gstreamer: [1.16](https://gstreamer.freedesktop.org/releases/1.16/)

Which, as far as I can tell, requires a 64-bit build of Windows. That is not working yet.

jaxxzer, 2019-07-18:
Made a separate issue regarding updating the deployment here https://github.com/mavlink/qgroundcontrol/issues/7591

DonLakeFlyer, 2019-12-24:
I think this is all good now. Windows has moved to 64 bit

closed 2019-12-24 by DonLakeFlyer