mavlink/qgroundcontrol #7526: Feature request 8 servos on heli setup page

twistedwings, 2019-06-14:
currently only 4 servos are supported on the new (and great) heli setup page.
On my controller Matek F405-Wing the first 2 servo slots are reserved for ESCs.

So the servo mapping looks like this in MP:

[image]

In qgroundcontrol it looks like this:

[image]

So motor3 and motor4 cannot be handled.

It would be great if the heli page could either handle 6 or 8 servos or figure out which servos are mapped to Motor1, Motor2, Motor3, Motor4 and show these.

DonLakeFlyer, 2019-06-14:
@twistedwings Do you know if the SERVO5-8_* parameters are something only available in more recent firmwares?

twistedwings, 2019-06-14:
My 1st ArduPilot was Copter 3.6.7.
But they should be there since Copter 3.4. 
[http://ardupilot.org/copter/docs/common-rcoutput-mapping.html](url)

DonLakeFlyer, 2019-06-14:
Can you verify tomorrow from a new Daily?

twistedwings, 2019-06-15:
I have tested this version:

[image]

I can't see any difference 

[image]

DonLakeFlyer, 2019-06-15:
Sorry the change hadn't been merged in yet. New daily build should be available in an hour.

twistedwings, 2019-06-15:
looks good, thanks alot!

[image]

closed 2019-06-15 by DonLakeFlyer