mavlink/qgroundcontrol #7538: Multi vehicle routes display issue

y27988, 2019-06-19:
in file "FlightDisplayViewMap"，The original：
MapItemView {
        model:  _mainIsMap ? _activeVehicle ? _activeVehicle.trajectoryPoints : 0 : 0
        delegate: MapPolyline {
            line.width:         3
            line.color:         _activeVehicle ?  _activeVehicle.trajectoryColor : "red"
            z:                  QGroundControl.zOrderTrajectoryLines
            path: [
                object.coordinate1,
                object.coordinate2,
            ]
        }
    }
it only display one vehicle route, and I modify the code like this:
    MapItemView {
        model: QGroundControl.multiVehicleManager.vehicles
        delegate: MapItemView {
            property var currentVehicle: object
            model:  _mainIsMap ? currentVehicle ? currentVehicle.trajectoryPoints : 0 : 0
            delegate: MapPolyline {
                line.width:         3
                line.color:         currentVehicle ?  currentVehicle.trajectoryColor : "red"
                z:                  QGroundControl.zOrderTrajectoryLines
                path: [
                    object.coordinate1,
                    object.coordinate2,
                ]
            }
        }
    }
in Linux OS, it can display all Vehicle route, but in windows 10, it can't display any Vehicle route.
Can anyone help me?

y27988, 2019-06-19:
anyone have other methods to display all Vehicle Route?

DonLakeFlyer, 2019-06-19:
You can't add Map* items to a Map dynamically using a repeater. The only way to do that is by calling Map.add* methods. If you look at something like QGCMapPolygonVisuals.qml you can see an example of adding things dynamically to a Map.

closed 2019-06-19 by DonLakeFlyer