mavlink/qgroundcontrol #7540: Record stream button placement (and keyboard shortcut)

dissident76, 2019-06-21:
ArduSub (BlueROV2)
Please add a "Record Stream" button somewhere else (and a keyboard shortcut) so the button can be pressed while displaying the telemetry Values. It takes too long to switch back and forth to the Video Stream menu every time I want to record a short video of marine life passing by -most of the times, action is over by the time it starts recording.

jaxxzer, 2019-09-09:
@dissident76 , this is one of our most common complaints after the update to 3.5. Thank you for filing a support ticket. Here is our proposal to address this issue:

1. Add a hotkey to start/stop local video recording
2. Add joystick button support to start/stop local video recording
3. Add visual feedback to the main UI to indicate when recording is **active**. While video recording is in progress, I suggest having a small red circle appear in one of these locations. When recording is not in progress, no circle is present.
[image]

@DonLakeFlyer @dogmaphobic will this approach work for you?

patrickelectric, 2020-06-15:
@DonLakeFlyer @dogmaphobic Is there any update in this case ?
There is a couple of complains about this and there is people that was not aware that QGC could record video since the actual position is terrible.

DonLakeFlyer, 2020-06-15:
So a major point of having the main instrument values in the toolbar now is so you don't have to keep the pages thing on the values page all the time. And you can instead keep it on the camera page if you like.

diveexplorer, 2023-01-21:
There would be a huge improvement if the record button would change color when you push it.. or record time would be visual underneath.. then it would be easier to see when you are recording and when you are not.

closed 2026-10-04 by github-actions[bot]