mavlink/qgroundcontrol #7544: Feature Request: Calibration Feedback Screen

thomasgubler, 2019-06-24:
Requested from @bkueng based also on feedback of @RomanBapst 
after the compass (and maybe accel/gyro too) calib, there should be some kind of window that shows you the new calibration values (and comparing against the previous ones).

DonLakeFlyer, 2019-06-24:
I had talked to @LorenzMeier About this a while back. QGC-ArduPilot already has support for this. If you look here under Compass Call you can see it: https://docs.qgroundcontrol.com/en/SetupView/sensors_ardupilot.html. I think it works really well and is understandable by mere mortals like me. This is supported through the MAVLINK_MSG_ID_MAG_CAL_REPORT in the ArduPilot mavlink dialect.

RomanBapst, 2019-07-22:
@DonLakeFlyer I guess what's missing here are the following parts:
1 )Px4 firmware needs to implement magnetometer quality checks.
2) We need to add a mavlink message which holds the mag cal information (not sure if there are existing ones we could use)
3) QGC needs to report the results to the user.

So we'll go ahead and do 1) and 2) before you can do anything.

DonLakeFlyer, 2019-07-22:
I think it would make a lot of sense to also cut over mag cal to mavlink message status reporting instead of status text messages at the same time. This would make it work better over noisy telemetry.

closed 2026-10-04 by github-actions[bot]