mavlink/qgroundcontrol #7547: Daily: Parameter Screen Empty

MaEtUgR, 2019-06-25:
## Expected Behavior
Show all the synced parameters under the "Parameters" tab.

## Current Behavior
[image]

## Steps to Reproduce:
1. Download latest daily
2. Connect to pixhawk with latest stable PX4 1.9.1
3. Try to change a parameter in the UI

## System Information
- QGC Version:
[image]

- Operating System: Windows 10 1809
- Flight Controller: pixhawk fmu-v5
- Autopilot: PX4 1.9.1 stable

dogmaphobic, 2019-06-25:
Found it. On it...

closed 2019-06-26 by dogmaphobic

MaEtUgR, 2019-07-17:
I upgraded to latest daily again and now it's all fine, thanks!

PR #7548 (2019-06-26): Revert ParameterEditor to QtQuick.Controls 1.x
In response to #7547

I had to revert `ParameterEditor` to QtQuick.Controls 1.x as it makes use of `ExclusiveGroup`, which no longer exists in newer versions of QtQuick.Controls. I will need to take a closer look at these and find a way around it.

Closes #7547 

P.S. These errors, when they occur, are all shown in the console:

```
qrc:/qml/SetupParameterEditor.qml:18:1: Type ParameterEditor unavailable
qrc:/qml/QGroundControl/Controls/ParameterEditor.qml:41:5: ExclusiveGroup is not a type
```

That would immediately tell me what went wrong.