mavlink/qgroundcontrol #7560: Reboot via QGC does not restore dropped RC connection

love-mj, 2019-07-03:
Quite often I have the RC control stop working after powering on the autopilot. It doesn't lose signal or indicate lost connection in QGC, it just stops responding to inputs from the RC radio. This only happens before arming, never after. It seems to not happen if I constantly move the sticks pretty regularly, but if left alone for a minute.

I'm not sure why this happens, but at the least if I could reboot via QGC and restore control that would be fine. That doesn't work however, so I have to open a hatch and physically unplug the power. Is there a known reason why the autopilot stops responding to input? If it's a big process to fix the issue, is there a way to restore it via software reboot for the short term?

love-mj, 2019-07-03:
An electrical reset can be done by either removing power or pushing the FMU reset button the side of Pixhawk next to the USB connector.
You can accomplish a firmware reset by using the CLI to reset the Pixhawk to factor parameters. You should save your existing parameters if you do not want them erased during a firmware reset.
Regards

DonLakeFlyer, 2019-07-03:
QGC isn't in control of any of this. You are going to need to talk with firmware/hardware setup folks.

closed 2019-07-03 by DonLakeFlyer