mavlink/qgroundcontrol #7567: Language selection issue

hslyu, 2019-07-08:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Language changes according to the selection

## Current Behavior
UI shows English regardless of the selection

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Windows 10, amd64 computer.
2. Download QgroundControl daily builds in official website.
3. After installation, select language
3. Nothing changes 

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10
- QGC Version: Development HEAD:1ec454401 2019-06-25 23:38:24 -0300
- QGC build: daily
- Flight Controller: Regardless
- Autopilot (with version): Regardless

## Detailed Description
Hi, I'm Korean translator of QgroundControl. For several months, QgroundControl language selection malfunctioned. I found that only approved translations are shown, but Korean translation is 100% approved. I did not test on another OS.

## Log Files and Screenshots
[image]

wuuhii, 2019-07-16:
The same to me!

dogmaphobic, 2019-07-18:
I can't "approve" the translations as I would have no idea if the translations are correct or not. That has to be done within Crowdin.

hslyu, 2019-07-18:
What I meant is that I already approved all of the Korean translation. I'm one of the proofreaders.

hamishwillee, 2019-07-18:
@gus It wasn't very clear, but the request is to update the translations for Korean used by QGC. Do you think we could also have a rebuild of the last stable release to include these?
[image]

DonLakeFlyer, 2019-07-18:
I'm working on a new Stable to go out in the next couple days. So whatever needs to happen for that needs to be changed in Stable_V3.5.1.

hamishwillee, 2019-07-18:
Cool. We decide 80% was our nominal cutoff for including a language translation in a stable release:
[image]

@gus @DonLakeFlyer , if I chase China, and Turkey to try get completion, when would be the cutoff for you to take a cut of the translations?

dogmaphobic, 2019-07-19:
>What I meant is that I already approved all of the Korean translation. I'm one of the proofreaders.

Ha! Got it. If it's ready at Crowdin, I just need to download it and integrate it. To _master_ though. Stable does not include Korean. That was only added after. The stable release still has the manual localization, with no language option. It loads the language based on the system settings, which proved to not work in many cases. Stable is only setup for Chinese and German.

gus, 2019-07-19:
@gus is a different Gus than you wanted, but I'm glad the right one replied
:)

Cheers,
@gus

dogmaphobic, 2019-07-19:
> @gus is a different Gus than you wanted, but I'm glad the right one replied :) Cheers, @gus

Another one? Who would think :P Cool. I was feeling all alone here :)

hslyu, 2019-07-20:
> Ha! Got it. If it's ready at Crowdin, I just need to download it and integrate it. To _master_ though. Stable does not include Korean. That was only added after. The stable release still has the [...]

Then, can I use language option normally in _daily build_ after you integrate available languages at Crowdin?

dogmaphobic, 2019-07-20:
> Then, can I use language option normally in _daily build_ after you integrate available languages at Crowdin?

Yes, I will take a look at it today.

dogmaphobic, 2019-12-26:
This is all done in master.

closed 2019-12-26 by dogmaphobic