mavlink/qgroundcontrol #7568: Second battery values not updating in widget UI.

xkam1x, 2019-07-08:
OS Version: macOS Mojave 10.14.5
QGC Version: 3.5.3 Stable Release
Flight Controller: Pixhawk 4 (Holybro)
Ardupilot: Copter 3.6.9 Stable Release

I have configured the Pixhawk to use both of its power ports for battery voltage and current monitoring. Pixhawk 4 uses ADC 0 and 1 on power port 1 for voltage and current respectfully and ADC 2 and 3 on power port 2 for voltage and current respectfully.

In the widget view I have enabled battery voltage and current and they update as expected, but voltage and current for battery2 stay at -1. See screenshot.

I have also verified that the Pixhawk is reporting correct values for both batteries by looking at the mavlink inspector. I see the following SYS_STATUS.voltage_battery and SYS_STATUS.current_battery both of these are read from power port 1 (ADC 0 and 1). In addition I have another message BATTERY2.voltage and BATTERY2.current_battery and they appear to be read from power port 2 (ADC 2 and 3). See screenshot.

In addition to updating the second battery values in widget view, it would also be helpful to have another icon for the second battery in status bar.

[image]
[image]

feemi, 2019-07-09:
We are experiencing same problem with Cube Pixhawk 2.1 hardware. ADC channels are 2-3 for main brick and 13-14 for backup brick.

DonLakeFlyer, 2019-07-16:
QGC does not support the BATTERY2 mavlink message which is ArduPilot specific. QGC expects battery 2 status to be updated using the mavlink generic BATTERY_STATUS message with BATTERY_STATUS.id set to different ids to indicate multiple batteries.

That said I thought that ArduPilot did support this BATTERY_STATUS multi-battery reporting. Can you use Analyze graphing (Menu/Widgets/Analyze) on BATTERY_STATUS.id to see if you are getting different values coming through?