mavlink/qgroundcontrol #7571: Pixhawk 2.1 Cube fmu-v3 usb connection problem

feemi, 2019-07-09:
## Expected Behavior
Autopilot should be connected via usb successfully.

## Current Behavior

We can not connect  to qgc from usb. This problem only occurs with usb connection. It connects successfully from telem1 uart via ftdi or similar. When we upload ardupilot and reflash px4 it looks fixed. But if change some parameters or update frame and reboot then problem reappears. Error screenshot is attached to issue.

Os recognizes usb device properly. Also QGC recognize on firmware upgrade tab. 

We are having this with various versions including v1.9.* We are having this problem in mac and windows and ubuntu with pixhawk 2.1 cube fmu-v3 board.

Note: pixhack5+ fmu-v5 is working properly.

## Steps to Reproduce:
1. If i reflash px4 firmware nothing changes.
2. If i flash to ardupilot it connects successfully to qgc.
3. Then i reflash px4 it also connects successfully to qgc.
4. But if i select airframe or change any parameter and restart then the issue raises again.

## System Information
When posting bug reports, include the following information
- Windows 10, Ubuntu 16.04, Ubuntu 18.04, Mac Mojave 10.14.5
- QGC Version: 3.5.3 and master
- QGC build: stable, build from source code
- Flight Controller: Pixhawk 2.1 Cube, fmu-v3
- Autopilot (with version): PX4 1.9.*, 

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
[image]

feemi, 2019-07-10:
In some cases formatting sd card fixes the issue but not every time. i ve seen similar old closed issues but i couldnt find the root cause.

jkflying, 2019-07-11:
Are you powering the FCU from the USB, or do you power it from batteries first before connecting the USB?

feemi, 2019-07-11:
We tried different cases.

For cube fmu-v3 it doesnt matter. Powering from usb or powering from battery doesnt change the result.
If cube connects properly with usb power. it connects also when powered from battery.

Pixhack 5+ fmu-v5. It doesnt connect when powering from battery first. It connects properly with only usb power.

jaxxzer, 2019-10-17:
I am experiencing a similar issue with pixhawk 1 on 3.5.4 and daily from 10/17. Here is what I see on windows:
[image]

DonLakeFlyer, 2019-10-17:
FYI: The Stable builds are screwed up right now due to a bad tag that was added. I'll kick off a new Stable today to fix it.

Don't know why daily doesn't work.

DonLakeFlyer, 2019-10-17:
@jaxxzer That feels like a driver problem. Try uninstalling/reinstalling drivers.

jaxxzer, 2019-10-17:
> Try uninstalling/reinstalling drivers.

How do I do that? (couldn't find anything in docs)

There are like 25 driver packages that look like they were installed. Do these belong to QGC? Is there any way to uninstall all of them at once?

[image]

DonLakeFlyer, 2019-10-17:
Those are old I think. The new one is three separate packages. Uninstall them all and then install a new daily and you'll get new drivers.

hamishwillee, 2019-10-17:
Remove "UAV Drivers". 

In addition, there is another package that is installed using a wizard called "Device Driver Installation Wizard" but I don't know what the install package is called.

hamishwillee, 2019-10-17:
Note, FYI apparently these aren't "essential" on windows 10. They set the com port names to the name of the actual board type (e.g. cube) rather than just "COM1". On earlier Windows they are essential.

DonLakeFlyer, 2019-10-18:
> Remove "UAV Drivers".

That is from the new drivers package. The old drivers list out individually like the screen shot. @jaxxzer You may have both installed. So check for both.

jaxxzer, 2019-10-18:
Ok, I'll be able to check next week.

Vourtsis, 2019-10-23:
I have the same issue on Pixhawk 4 fmuv5. On top of that Pixhawk 4 overheats when I attempt the USB connection.

jaxxzer, 2019-11-06:
I'm sorry, my problem pertains to firmware flash in particular, and is unrelated to this issue. I've moved my comments here: https://github.com/mavlink/qgroundcontrol/issues/7981

SMenne, 2019-11-12:
I solved that Problem with using Mission Planner / New Firmware / Load Customer Software 
Then I took the PX4 Software "px4fmu-v3_default.px4"

byoruk, 2020-01-03:
> I solved that Problem with using Mission Planner / New Firmware / Load Customer Software
> Then I took the PX4 Software "px4fmu-v3_default.px4"
> After that no Problems anymore with QGC

This is worked for me to connecting QGC, too. But I want to ask is there anyone work on Matlab px4 support package in Windows. When I work with that I noticed Port Configuration changed in Device Manager. In first, connecting cube to USB port, Device Description shown "ProfiCNC CUBE F4 FMU (COM3)" after 3 sec. description changed and shown "Legacy FMU(COM6)".  Because of that issue, Matlab confused. And can't connect to device.  How can I handle this? Is there any sugggestion?

Hayden8135, 2020-01-08:
> I solved that Problem with using Mission Planner / New Firmware / Load Customer Software
> Then I took the PX4 Software "px4fmu-v3_default.px4"
> After that no Problems anymore with QGC

Thank you so much. but I still wondering why it was problem about USB driver?? (-_-) still curious..
Actually, I want to find this issue more easier then I tried install a lot of things virtual usb port for this issue. 
but anyway, I solved by update from px4fmu-v3_default.px4

Hayden8135, 2020-01-29:
> I solved that Problem with using Mission Planner / New Firmware / Load Customer Software
> Then I took the PX4 Software "px4fmu-v3_default.px4"
> After that no Problems anymore with QGC

Sorry, mate 
would you mind replay for me link? I forgot how to go there.
please thanks

closed 2026-10-04 by github-actions[bot]