mavlink/qgroundcontrol #7572: Parameter setting is not applied when reboot a vihicle

start0821, 2019-07-10:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
I need to set `SYS_COMPANION` in `SYSTEM PARAMETER` to Companion Link(921600) and reboot the vihicle to apply the setting. But when I reboot the vihicle, it come back to default value `disable`. What can I do?

## Current Behavior
I need to set `SYS_COMPANION` in `SYSTEM PARAMETER` to Companion Link(921600) and reboot the vihicle to apply the setting. But when I reboot the vihicle, it come back to default value `disable`. 

## System Information
When posting bug reports, include the following information
- Operating System: window10
- QGC Version: 3.5.3
- QGC build: stable (maybe?)
- Flight Controller: Pixhawk1
- Autopilot (with version): PX4 1.9.3

## Detailed Description
I need to connect a pixhawk to raspberry pi using `TELEM2`, so I set `SYS_COMPANION` in `SYSTEM PARAMETER` to Companion Link(921600) and reboot the vihicle to apply the setting. But when I reboot the vihicle, it come back to default value `disable`. What can I do?  

other `SYSTEM` parameter does not have this problem. 

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
+ before rebooting the vihicle
[image]

+ after rebooting the vihicle ( parameter value comes back to default)
[image]

- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

feemi, 2019-07-10:
Hi,

Sys_companion parameter is deprecated as it seen. So you should use mavlink parameters

There is MAV0_ and MAV1_ and MAV2_ options
Usually MAV0 is used for ground control station telemetry.

You can assign MAV1_ for onboard communication. You should select the interface on MAV1_CONFIG parameter. In your case i assume it would be TELEM2, then you can select MAV1_MODE to onboard.

[image]

Note: I guess you meant 1.9.2 i guess because there is no released version as 1.9.3.

start0821, 2019-07-10:
> Hi,
>
> Sys_companion parameter is deprecated as it seen. So you should use mavlink parameters
>
> There is MAV0_ and MAV1_ and MAV2_ options
> Usually MAV0 is used for ground control station telemetry. [...]

Thank you for replying !!
I edit the QGC version to 1.9.2 and I'll try again!

feemi, 2019-07-10:
Hi @DonLakeFlyer I guess you can close this issue.

closed 2019-07-10 by dogmaphobic