mavlink/qgroundcontrol #7578: QGC(Debug Edition) Crashed as Soon as It Started Running

wuuhii, 2019-07-15:
### Ground station operating system
windows 10 64bits professional edition
### QGroundControl version and build
Building from master(with Qt 5.11.3 and vs 2015)
### Exact steps to reproduce the problem
Running the application after building finished, the bug will occur. (The bug will occur when build with debug edition, release edition has no the bug.)
### Screenshot
[image]

DonLakeFlyer, 2019-07-16:
@dogmaphobic This is most likely the thermal camera rendering crash I was telling you about a while back

wuuhii, 2019-07-18:
Will this bug be fixed first? I can not debug my program now.  @dogmaphobic

DonLakeFlyer, 2019-07-18:
Working on it...

wuuhii, 2019-07-22:
nice!