mavlink/qgroundcontrol #7580: Microhard issues.

edwinhayes, 2019-07-15:
I've got some concerns about the way that 'support' for Microhard radios as been added in master (PR #7291 and related).

Mostly, it changes settings on both the local and remote radio every time it attempts to connect (`MicrohardSettings.cc -> _readBytes()`) - namely the radio encryption key (via the `AT+MWVENCRYPT` command).

This is pretty insane.  There is basically no documentation or other indication to the user that this is going to happen, and if one of any number of thing goes wrong, could result in becoming permanently locked out of an aircraft in flight, including both telemetry and primary flight controls.

1. If all you want to do is receive RSSI and other configuration information, there is no reason to be logging into the radio as 'admin', which it's currently hard coded to do.  Should instead take a  username field so that RSSI querying can be performed by a user with reduced permissions, and only use a user with elevated permissions when required for changing configuration settings.
2. Changing the encryption key on the remote end doesn't make sense in the majority of use cases.
3. Providing a configuration tool which _only_ manages the encryption key is basically a waste of time, since to connect a pair of radios, you also need to make sure all the other relevant radio settings match up, so this is likely to cause confusion with accomplishing much.
4. If you're going to manage the configuration of radio settings, you need to perform this via a proper state machine which always makes changes to the remote end _before_ making the matching change on the local end, and rolls back the changes in a coherent manner if something goes wrong.

Basically, all this seems to do is fetch RSSI values, but does so in an unnecessarily dangerous way.

edwinhayes, 2019-07-15:
(I realise most devs probably won't have the equipment to test stuff like this; we run MH radios, so we're happy to do it.)

MatejFranceskin, 2019-07-16:
Hi. I think you're right. The configuration for setting the encryption was needed in a particular use case and introduces potential problems. As well as point 1. I will make some changes and get back to you.

MatejFranceskin, 2019-07-19:
Point 1 is addressed in https://github.com/mavlink/qgroundcontrol/pull/7601
2. Remote encryption key is never changed by QGC. Only local is changed.
3. There is a new project in development called "pairing manager" that will allow exchange of information like encryption key between UAV & QGC via physical contact (NFC). For this case setting of encryption key will be needed.
4. Currently we don't and don't need to set remote encryption password but otherwise you're right. If we're going to implement this we have to be careful.

closed 2019-07-25 by MatejFranceskin