mavlink/qgroundcontrol #7581: Telemetry RSSI indication inconsistency.

edwinhayes, 2019-07-15:
The Telemetry RSSI widget (`TelemetryRSSIIndicator.qml`) hard codes the RSSI to be displayed with " dBm" afterwards.  The value itself is reported as a positive integer, so you set something like:

[image]

But 46dBm would be pretty hot; the actual value is -46dBm.  However, the MAVLink RADIO_STATUS message which populates this widget encapsulates RSSI as a `uint8_t` with "device dependent" units.

So either the widget should just display the positive integer value without denoting any particular units, or it should prefix the dBm value with a negative sign (I'd prefer the latter, since I'd argue than any sane radio should be reporting its RSSI in dBm, but this would be tacitly accepting that some 'other' radios which report RSSI as a percentage or similar would read incorrectly in the widget).

DonLakeFlyer, 2019-07-16:
I think there was some big discussion on this a while back since the mavlink spec was clear as mud on it. @dogmaphobic would know more

hamishwillee, 2019-07-16:
If this comes from RADIO_STATUS then @edwinhayes is correct - this is "device dependent units" (0-254, with 255 indicating not used). 
Of course not every system may be publishing this correctly.

We standardised on this for all RSSI values fairly recently. The only exception is [CELLULAR_STATUS](https://mavlink.io/en/messages/common.html#CELLULAR_STATUS)

Discussions:
- https://github.com/mavlink/qgroundcontrol/issues/7013
- https://github.com/mavlink/mavlink/issues/1027

edwinhayes, 2019-07-16:
@hamishwillee - Yes, it comes from `RADIO_STATUS`, the widget is hidden without.

Yeah, I recall mavlink/mavlink#1027.   I was happy with the outcome, except I noted that I thought a further improvement would be for QGC to have a pair of configuration values for transforming the raw value to a linear % representation, to address how to display a device dependent value in a user readable form.

But absent that, the current labelling is inconsistent.

closed 2026-10-04 by github-actions[bot]