mavlink/qgroundcontrol #7585: Vehicle setup qml timers remain active after navigating to fly view

jaxxzer, 2019-07-17:
As noted in #7275 :
> There is another bug involved, and that is that the code/timers in the vehicle setup qml pages keeps running after the fly view is selected. So if motor test page is open in the background (while fly view is active), then it is sending motor test commands after you arm the vehicle. 

This is confirmed to occur on master for the motor setup page, I haven't investigated yet to see if any other setup pages are affected.

DonLakeFlyer, 2019-07-17:
The pages from each view (Settings, Setup, Plan, Fly) continue to run even if they are not being shown.

DonLakeFlyer, 2019-12-12:
The pull reference is the first step to providing a mechanism to fix this. When done it will provide a system for a page to prevent a view switch. Hence if you are in the middle of a calibration or an active motor test you won't be able to switch views. Preventing arming will be a little trickier. But view switch is a worse problem I think.

DonLakeFlyer, 2019-12-12:
> I haven't investigated yet to see if any other setup pages are affected.

They are. Things which are say calibration relation with a start/stop to calibration will be affected.

DonLakeFlyer, 2019-12-12:
@jaxxzer Once #8113 goes through you'll have the mechanism to fix your motor test page.

jaxxzer, 2019-12-17:
ok! :clap:

DonLakeFlyer, 2020-02-08:
@jaxxzer This should get fixed before I cut a stable which is soon

DonLakeFlyer, 2020-02-21:
Time is running out to get this fix into stable.

Williangalvani, 2020-02-21:
That was fixed in my latest PR, you can close it!

closed 2020-02-21 by DonLakeFlyer