mavlink/qgroundcontrol #7586: Feature Proposal - Home Heading / COG Angle / Next WP Heading in Compass

tilaktilak, 2019-07-17:
We worked on some indicators in the compass widget to show :
- Home Heading
- COG Angle (ie. The GPS Heading from trajectory)
- Heading of the next Waypoint

We also added the ability to lock the North up (The red vehicle arrow locked up).

From our first implementation it looks like this :
[image]

Settings Page :
[image]

Here the current implementation : https://github.com/mavlink/qgroundcontrol/compare/master...Aeronavics:compass_home_n_COG

Wanted to have some feedback if it worth integrating those features in the mainline.

DonLakeFlyer, 2019-07-17:
All seems valuable to me. I would change the settings text to be just "Show" from "Show/Hide".

@antiheavy Opinions?

edwinhayes, 2019-07-17:
If people think this is too busy, we could condense the three show/hide checkboxes into one?  If someone wants COG but not home, they can rebuild themselves? (But personally, I don't mind long lists of configuration options).

Also, suggest changing "Lock North up on Compass" to "Lock Compass 'North-Up'".  Oh, hang on, the usual default behaviour is for North-Up, so the checkbox should be "Lock Compass 'Nose-Up'" so that  having all the boxes unchecked delivers the existing 'normal' QGC experience?

edwinhayes, 2019-07-17:
Oh, also remove the word "angle".  And change "Show Home heading to Compass" to "Show Heading to Home on Compass" so that it's consistent with "Show Heading to Next Waypoint on Compass".

Antiheavy, 2019-07-18:
I like this.  +1 vote from me.

Antiheavy, 2019-07-18:
> All seems valuable to me. I would change the settings text to be just "Show" from "Show/Hide".

correct.  checkbox options like that should just be "Show"

Antiheavy, 2019-07-18:
>  we could condense the three show/hide checkboxes into one?

seems okay.  maybe "Show extra heading indicators on compass"

How does a new user know what indicator means what thing?

edwinhayes, 2019-07-18:
> How does a new user know what indicator means what thing?

 @tilaktilak can update the documentation page online to explain?

tilaktilak, 2019-07-18:
Sure we can add those infos in the manual

DonLakeFlyer, 2019-07-18:
I would:
* Include Home as showing always
* Having a single "Show additional heading indicators in compass"
* I would position all indicators to the edge of the circle (like home in image). Then zorder from lowest the highest: COG, Home, Waypoint.

edwinhayes, 2019-07-18:
> I would position all indicators to the edge of the circle (like home in image). Then zorder from lowest the highest: COG, Home, Waypoint.

We might try some different arrangements to evaluate for HMI effectiveness?

Also, one thing that Pierre didn't mention, which we probably want people's thought's on; currently the home marker disappears when the aircraft is within a short distance from home, and similarly the COG indicator disappears when the aircraft is hovering; this avoids the indicators spinning around the rose when the magnitude of each vector is close to zero.  This behaviour and the associated thresholds could be fixed or configurable.

DonLakeFlyer, 2019-07-18:
> currently the home marker disappears when the aircraft is within a short distance from home, and similarly the COG indicator disappears when the aircraft is hovering; this avoids the indicators spinning around the rose when the magnitude of each vector is close to zero.

Makes a ton of sense

tilaktilak, 2019-07-18:
> I would:
> 
>     * Include Home as showing always
> 
>     * Having a single "Show additional heading indicators in compass"
> 
>     * I would position all indicators to the edge of the circle (like home in image). Then zorder from lowest the highest: COG, Home, Waypoint.

I'll implement this to see how it goes and post a screenshot.

`This behaviour and the associated thresholds could be fixed or configurable.` :

- There is already a filter on HeadingToHome value :   

    Value not available if we are closer than 1m [SEE HERE](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L3774) 
   

- I added a filter to hide COG if ground speed is < 0.5m/s 

    As @edwinhayes could be a parameter instead of a hard-coded value but we may want to do it also for the 1.0m value in the HeadingToHome update

- For the NextWP, it's setted ot 5m which may need to be align to 1m ( Same as for Home heading)
[Code is Here](https://github.com/Aeronavics/qgroundcontrol/blob/24c0133d6b14838a86631303f206ea0b5182fe8d/src/Vehicle/Vehicle.cc#L3798)

DonLakeFlyer, 2019-07-19:
I knew I forgot something: Can you plumb the values for these headings through Vehicle is Facts? That way the will also be available for display in the instrument panel as values. If the values are no available then set them to NaN.

tilaktilak, 2019-07-19:
Actually I did that but I missed the add_Fact part so I fixed it, now it's good.
Had to shorter the text a little bit to fit the value setting page :
[image]

I also change the diameter of the indicators and set them in the order from lowest the highest: COG, Home, Waypoint.

[image]

DonLakeFlyer, 2019-07-19:
Time to submit as a Pull

tilaktilak, 2019-07-21:
I may propose changes in the color used just before sending the PR : 
Changed the Next WP indicator to green to match the next WP color on the map.
Changed the COG indicator to blue (something not green).
Here a draft :
[image]

What do you think ?

edwinhayes, 2019-08-02:
Maybe also want to consider including space for collision avoidance information.  Refer to this image of the corresponding screen in MP:

[image]

DonLakeFlyer, 2019-08-13:
Collision avoidance UI is already being discussed in a separate issue: https://github.com/mavlink/qgroundcontrol/issues/7670

@tilaktilak I think this issue can be closed now?

closed 2019-08-13 by tilaktilak