mavlink/qgroundcontrol #7589: Landing pattern error with ardupilot

kikislater, 2019-07-18:
## Expected Behavior
Upload mission with landing pattern success to ardupilot fw

## Current Behavior
Upload mission with landing pattern to Pixhawk arduplane 3.9.8 fait with this error : MAV_MISSION_INVALID_PARAM5_X

[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect to arduplane 
2. Plan a landing pattern 
3. Upload to board
4. Watch error

## System Information
When posting bug reports, include the following information
- Operating System: Archlinux
- QGC Version: 3.5.3
- QGC build: stable
- Flight Controller: Pixhawk 1, SITL in Vagrant
- Autopilot (with version): Arduplane 3.9.8, Arduplane 3.10.0-dev

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

DonLakeFlyer, 2019-07-18:
This should be fixed in daily builds. I had thought I brought it across to Stable as well. Let me looka gain.

kikislater, 2019-07-18:
Yes ! Works fine. 
Out of curiosity, where is related PR ?

DonLakeFlyer, 2019-07-18:
There are various fixes. Discussion is here with links: https://github.com/mavlink/qgroundcontrol/issues/6783

holden-zenith, 2022-02-10:
This should be closed

closed 2022-02-11 by DonLakeFlyer