mavlink/qgroundcontrol #7599: [FeatureRequest] Display Companion Status on QGC

Jaeyoung-Lim, 2019-07-19:
Flying avoidance more often these days, having a companion status display would be much convenient on checking if the companion computer is running properly.

Currently only a text message (`Avoidance system Connected` / `Avoidance System Lost`) is displayed, and results as the following. 
- The message can be lost sometimes when you relaunch qgc
- The message floods the message box as it repeatedly displayed as connected / disconnected whenever there is a glitch in the network
- We need to wait for the companion to be up and running before using avoidance, where the only indicator is `Avoidance System Connected`

Suggestion:
It would be great if the companion status can be displayed in the top bar with a color code showing the status of the companion. This would greatly enhance the visibility of the status of the companion.

LorenzMeier, 2019-08-07:
@baumanta How are we looking in terms of specs for this? We need to have a clear idea which fields need to go in there.

hamishwillee, 2019-08-07:
@LorenzMeier Spec appears to have stabilised: https://github.com/mavlink/mavlink/pull/1186

It needs to be rebased, but IMO is good to be merged.

jkflying, 2019-08-21:
This is missing a few things now to get to a basic working point

- ROS node which gathers data and sends it to MAVROS - https://github.com/Auterion/system_monitor_ros - still in early stages
- MAVROS plugin to send the new ONBOARD_COMPUTER_STATUS message - https://github.com/mavlink/mavros/pull/1286 - waiting on MAVLink release to merge
- QGC integration to show populated system, maybe similar to GPS status? @yasendimov maybe you have some idea of how it should look?

Jaeyoung-Lim, 2019-11-16:
PX4 Firmware now supports handling and logging the onboard computer status messages as of https://github.com/PX4/Firmware/pull/13169

closed 2026-10-04 by github-actions[bot]