mavlink/qgroundcontrol #7600: QGC crashing and not dying properly

baumanta, 2019-07-19:
I am using the QGC App Image 3.5.0 under Ubuntu 16.04.
Mainly I use it in combination with the PX4 SITL setup. The App crashes quite often for me but I cannot make out a specific reason or action linked to it. Every time it crashes and sometimes also when I close it on purpose, the App does not shut down properly. There are two processes still running which I have to SIGKILL or otherwise I cannot restart the App anymore. 

I'm not sure how I can help best to debug this. But please let me know if I can provide any log/screenshot or other information.

baumanta, 2019-07-19:
FYI @MaEtUgR

DonLakeFlyer, 2019-07-19:
First step is to use the latest Stable version which is 3.5.3

baumanta, 2019-07-23:
I have the same issues on 3.5.3

DonLakeFlyer, 2019-07-25:
> There are two processes still running which I have to SIGKILL

There are two QGroundControl processes?

DonLakeFlyer, 2019-07-25:
@tcanabrava Do you know is there some way in Linux to capture stack traces by doing some OS magic?

tcanabrava, 2019-07-25:
Yes, I’ll send a patch for that.

DonLakeFlyer, 2019-07-25:
Also this doesn't happen to be on a Lenovo Laptop does it?

baumanta, 2019-07-26:
yes there are two processes, they show up as:
`/home/tanja/Programs/QGroundControl.AppImage`
`/tmp/.mount_2jcZdi/QGroundControl`

and yes it is a lenovo laptop, model T470p

DonLakeFlyer, 2019-07-26:
Are you streaming video in QGC?

DonLakeFlyer, 2019-07-27:
Also make sure you your intel integrated graphics drivers are completely up to date.

baumanta, 2019-07-29:
no, I do not stream video. I have an intel i915 and a nvidia driver running on my laptop. What information would you need about those?

DonLakeFlyer, 2019-07-29:
Ok, so you should try the 3.5 stable which was just released and has a crash fix in it for gstreamer when you aren't using video. It may fix the problem. But it is more likely to be a known problem with Intel OpenGL driver support which we are trying to find a workaround for. For some people updating the i915 drivers to the latest fixed it. The other thing that works is forcing to always use the nvidia driver even when on battery.

baumanta, 2019-07-30:
ok if that is a known issue, I will close this. Just wanted to make you aware of it and see if I could help anything for degugging

DonLakeFlyer, 2019-07-30:
I'd rather keep this open for now. So when we figure it out we know who can test it out to verify.

Have you tried 3.5?

dogmaphobic, 2019-07-30:
@baumanta, Stefan is having similar issues on CentOS. "3.5" is the "Stable" release of QGC. I don't think anybody in the office uses it.

booo, 2022-09-15:
Hopefully resolved in newer issues of QGC.

closed 2022-09-15 by booo