mavlink/qgroundcontrol #7602: Revisit usage of SYS_STATUS and BATTERY_STATUS values

DonLakeFlyer, 2019-07-19:
Currently a set of battery 1 fact values come from SYS_STATUS and some from BATTERY_STATUS. There is the possibility that the definition of what SYS_STATUS actually sends may change. It may make more sense for all values to come from BATTERY_STATUS if that message is sent by the vehicle.

hamishwillee, 2019-07-21:
> It may make more sense for all values to come from BATTERY_STATUS if that message is sent by the vehicle.

It _definitely_ does. In theory the GCS can report anything about individual batteries, and also aggregate batteries (ie if there are dual batteries it could report the cumulative status). Using SYS_STATUS cannot support this.

hamishwillee, 2019-07-21:
Note also there are also SMART_BATTERY_ messages that are only sent for smart batteries. These, like BATTERY_STATUS, are sent individually for multiple batteries.

DonLakeFlyer, 2019-12-24:
@hamishwillee Was there ever any conclusion on this?

hamishwillee, 2020-01-06:
@DonLakeFlyer No. Too much arguing on detail. I will get back to this, but it is not my biggest priority right now.

Current state is that `SYS_STATUS` battery information usage is ambiguous and arbitrary:
- For PX4 it is the battery remaining % of the lowest of up to 2 batteries (I don't know how voltage/current values are used).
- For ArduPilot it is the value of one real battery (which battery determined by parameter).
- For Michael DuBreuil GCS it is "just there as a spare debugging value that gets logged".

FWIW I still argue that `SYS_STATUS` should not be used at all. The GCS can provide a much more consistent experience using SMART_BATTERY_  and BATTERY_STATUS messages.

DonLakeFlyer, 2020-02-08:
I tweaked this more to fix battery 2 reporting. Which makes it as good as it can be given current state.

closed 2020-02-08 by DonLakeFlyer