mavlink/qgroundcontrol #7605: version 9fd5cbc: CameraManager VideoStop broken

olliw42, 2019-07-20:
title says it

dogmaphobic, 2019-07-20:
Can you expand on that a bit? Is it not sending anything? Sending the wrong thing? Something else?

olliw42, 2019-07-20:
when in video mode, pressing the red round button sends a startVideo() and the button goes square as usual, but the button gets GRAY and doesn't remain red, from then on it's not possible to get QGC emit a stopVideo() by clicking it. In photo mode it works.

Btw, unrelated, but just to have mentioned it, there are also plenty of gray transparent square widgets popping up all the time when using the camera manager gui.

olliw42, 2019-07-31:
issue persists in 6ebae0a

olliw42, 2019-10-05:
new issue raised to bring it up, so closed this

closed 2019-10-05 by olliw42