mavlink/qgroundcontrol #7607: about localization

david6096, 2019-07-22:
The following template is for Issue/Bug reporting only. https://docs.qgroundcontrol.com/en/Support/Support.html#github-issues

For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

If you are using an old stable release please upgrade first and verify your issue before reporting.
----

## Expected Behavior
Please tell us what should happen as detailed as possible.

## Current Behavior
Please describe what happens instead of the expected behavior.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: [e.g. iOS 12.1, Windows 8, macOS Mojave 10.14.1, Ubuntu 14]
- QGC Version: [e.g. 3.4.4]
- QGC build: [e.g., daily, stable, self-built from source, etc...]
- Flight Controller: [e.g., Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.]
- Autopilot (with version): [e.g., PX4 1.8.1, Ardupilot Copter 3.6.1]

## Detailed Description
Provide further details about your issue/bug.

## Log Files and Screenshots
- [QGC Console Logs](https://docs.qgroundcontrol.com/en/Support/Support.html#reporting-bugs)
- Autopilot logs when available (post a link).
- Screenshots of QGC to help identify the current issue/bug behavior.

david6096, 2019-07-22:
in QGCApplication.cc,there is the following source code:
 void QGCApplication::setLanguage()
{
  ...
  if(_QGCTranslator.load(_locale, QLatin1String("qgc_"), "", ":/i18n")) {
        QLocale::setDefault(_locale);
        _app->installTranslator(&_QGCTranslator);
    } else {
        qDebug() << "Error loading application localization for" << _locale.name();
    }
  ...
}

   but there is not i18n resource group  in qgcresources.qrc, I wondered is it a bug?
   thanks.

dogmaphobic, 2019-07-22:
That's Qt's own, builtin localization resource path. It's created automatically here:
https://github.com/mavlink/qgroundcontrol/blob/master/qgroundcontrol.pro#L1226-L1227

closed 2019-07-22 by dogmaphobic

david6096, 2019-07-22:
but when I run the qgroundcontrol on ubuntu 16.04,the console logs the following msg:
error loading application localization for "zh_CN"

datwwe, 2020-08-12:
> but when I run the qgroundcontrol on ubuntu 16.04,the console logs the following msg:
> error loading application localization for "zh_CN"

I'm having the same problem. Have you solved this yet?
Any help are appreciated!