mavlink/qgroundcontrol #7612: Reposition vs Mission Pause

Stifael, 2019-07-23:
For both `GoTo` and for `Mission-Pause` the mavlink-msg `MAV_CMD_DO_REPOSITION(192)` is used. However, there is already a dedicated msg for pausing/continuing a mission called `MAV_CMD_DO_PAUSE_CONTINUE (193 )`. To be able to distinguish between `GoTo` and `Pause` also within the log, it would be beneficial to use 193 for misisson-pause instead of 192.

Stifael, 2019-07-25:
@DonLakeFlyer do you have any objection against my proposal to use MAV_CMD_DO_REPOSITION for repositioning only and MAV_CMD_DO_PAUSE_CONTINUE  for mission pause? if yes, then I can make the corresponding changes in upstream PX4

DonLakeFlyer, 2019-07-25:
The problem is that if firmware doesn't already support MAV_CMD_DO_PAUSE_CONTINUE then QGC gets into a mess of needed to know which version supports it or not.

closed 2026-10-04 by github-actions[bot]