mavlink/qgroundcontrol #7614: Joystick calibration does not survive QGC restart

MaEtUgR, 2019-07-24:
## Expected Behavior
Joystick calibration needs to get done once, gets saved and persists a restart of QGC or the device.

## Current Behavior
After #7595 the calibration is gone again after restarting QGC.
- I only have this on one specific combined tablet and joystick device. On my computer with an xbox controller and SITL it works.
- When I restart QGC and calibrate again because I have to the joystick still doesn't work but I have to calibrate once more and then it starts working again.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Connect to vehicle
1. Got to joystick tab and calibrate it
1. verify it all works fine
1. close QGC gracefully by pressing the X in the top right corner
1. Start QGC again and see the Joystick tab being red and needing recalibration

## System Information
When posting bug reports, include the following information
- Operating System: Windows 10 1809 (both tests)
- QGC build: Daily `HEAD:2aa69014f 2019-07-23 18:55:35 -0700` (both tests)
- Flight Controller: pixhawk 4 (SITL working test test)
- Autopilot (with version): PX4 master

FYI @dogmaphobic. Please tell me if I need to provide more information or conduct certain specific tests.

dogmaphobic, 2019-07-24:
I could not reproduce this. Once calibrated, it stays calibrated. Saving the calibration is the last step in the process, when you hit "Next" at the very last calibration phase. The button then reverts to "Start" allowing you to re-calibrate if desired.

MaEtUgR, 2019-07-27:
Yes, it works as expected on my Windows computer but fails on the tablet. I should probably check if other settings get saved properly to make sure it's not a general problem and then try to activate some debug logging. Is there a difference when I run the release AppImage or a self-compiled release using the `qgroundcontrol-run.sh` script?

dogmaphobic, 2019-07-27:
I'm pretty sure this was solved here:
https://github.com/mavlink/qgroundcontrol/pull/7622

I found a scenario where this would be the case. I wasn't aware the code in QCG treats "Gamepads" differently. You can open it again if it's still not solved.

closed 2019-07-27 by dogmaphobic

MaEtUgR, 2019-07-29:
Good find, that would explain why it was working on my PC but not on the micronav. I don't have the problem anymore with the latest version. Thanks!