mavlink/qgroundcontrol #7615: Wrong text color on close popup

FlavioTonelli, 2019-07-24:
On current master when QGC is closed with active connection the warning windows has a wrong text color.
It plots black on black(see screenshot)
[image]

dogmaphobic, 2019-07-24:
That's a system dialog. Not something coming from QGC. That is, not something QGC is drawing on its own. That seems to be an issue with Qt and macOS' "Dark Mode".

FlavioTonelli, 2019-07-25:
@dogmaphobic thank you for your feedback. I didn't notice it was created by macOS itself.
Isn't it fixable then?

dogmaphobic, 2019-07-25:
I don't know yet. I need to do some research. I know Qt had issues with "dark mode" but it was supposed to have been fixed in Qt 5.12. I'm running 5.12 and I still can see the same issue.

dogmaphobic, 2019-12-26:
This appears to have been fixed by Qt

[image]

closed 2019-12-26 by dogmaphobic