mavlink/qgroundcontrol #7616: Camera_capture msg type mismatch

FlavioTonelli, 2019-07-24:
PX4 camera_capture uorb message has a timestamp field declared as a uint64 timestamp.
QGC instead save the imported data inside a structure GeoTagWorker::cameraFeedbackPacket with field timestamp declared as double.
This lead to a mismatch between the saved uorb data and the QGC one. 
This is a bug. isn't it?

DonLakeFlyer, 2019-07-24:
The values stored in those two fields are in seconds translated from the log values, hence double. That seems fine to me?

FlavioTonelli, 2019-07-24:
I can't get your point. Neverless of the unit of measure we are implicitly converting a uint64 variable into a double. Isn't it wrong?
The log value and the QGC does not match.(See screenshot) 
Am I missing something?
[image]

DonLakeFlyer, 2019-07-24:
> The log value and the QGC does not match

I understand that. But as far as I understand the cameraFeedbackPacket data structure in QGC is only used internally to QGC. Why do they need to match?

DonLakeFlyer, 2019-07-24:
They don't match from a data structure standpoint. But they do match from a semantic standpoint as far as representing the same amount of time with the internal QGC value being representing in floating point seconds.

FlavioTonelli, 2019-07-24:
Ok now I get how was it is suppose to work.
Yes they both represent time you are right. What you lose is a 1 to 1 connection with the drone log and you probably lose some time accuracy.
Which are the Pro of such approach?

DonLakeFlyer, 2019-07-24:
> Which are the Pro of such approach?

No idea. You would have to ask @birchera who wrote that code.

birchera, 2019-07-24:
As far as I can see the cameraFeedbackPacket were introduced 2 years ago by @mhkabir, maybe he can answer your question.

DonLakeFlyer, 2019-07-25:
Discussion can continue, but I'm closing issue since it's not a bug.

closed 2019-07-25 by DonLakeFlyer