mavlink/qgroundcontrol #7621: PX4 waypoint - Remove MIS_YAWMODE check(s)

bresch, 2019-07-26:
`MIS_YAWMODE` does not exist anymore on the firmware side and has been kind of replaced by `MPC_YAW_MODE`. I mean "kind of" because the waypoint heading will always override the behavior dictated by `MPC_YAW_MODE`.
This means that the "heading" option of a waypoint should only check for the type of vehicle (has to be a multorotor) and not for a specific parameter in order to show-hide the widget.

It might be that that parameter is also checked somewhere else and we should make sure to remove those checks.

## Expected Behavior
[image]

## Current Behavior
[image]

## Steps to Reproduce:
- Open QGC
- Go to the mission planning tab
- Add a waypoint (it will show heading option)
- Connect to a vehicle (SITL or real one)
- Add a waypoint (it will not show the heading option)

## System Information
Tested on Ubuntu 16.04 and Andoid tablet with current QGC

DonLakeFlyer, 2019-07-26:
@dogmaphobic Are you working on this. If not I can do it.

dogmaphobic, 2019-07-26:
I can do it quickly. I just finished the joystick stuff...

closed 2019-07-26 by dogmaphobic

bresch, 2019-07-29:
Thank you guys!

PR #7623 (2019-07-26): Show heading on waypoint if multi-rotor.
In response to #7621 

For PX4, don't hide heading if vehicle is multi-rotor. No longer checks for a parameter.

Closes #7621