mavlink/qgroundcontrol #7635: Linux AppImage fix link libts

tilaktilak, 2019-07-28:
In [create_linux_appimage.sh](https://github.com/mavlink/qgroundcontrol/blob/1f283ad0dda59f06d3df9cdebbf02a7ed42453e1/deploy/create_linux_appimage.sh#L61) this link is dead.
We could replace this line by : ` wget -c --quiet "http://old-releases.ubuntu.com/ubuntu/pool/universe/t/tslib/libts-0.0-0_1.0-11_amd64.deb"

DonLakeFlyer, 2019-07-29:
Is this something really new? I just rebuilt stable 3.5 and the app image worked for me when I ran it.

DonLakeFlyer, 2019-07-29:
Also if this is currently failing if you could help to fix the script such that it fails the build if the wgets fail that would be awesome.

tilaktilak, 2019-07-29:
working on it at the moment,
I probably go for something like that : 
```
# exit when any command fails
set -e

wget -c --no-verbose http://ftp.us.debian.org/debian/pool/main/u/udev/udev_175-7.2_amd64.deb
wget -c --no-verbose http://ftp.us.debian.org/debian/pool/main/s/speech-dispatcher/speech-dispatcher_0.8.8-1_amd64.deb
wget -c --no-verbose http://ftp.us.debian.org/debian/pool/main/libs/libsdl2/libsdl2-2.0-0_2.0.2%2bdfsg1-6_amd64.deb
```

I added set -e to exit in case of an error
I changes --quiet to --no-verbose so it does not print everything but the errors   

And that make me realized that a lot of the wget result in a 404 error, that's mainly because those packages are old (from debian wheezy). 
It probably worth to have a look of the all script to update the dependencies.    

Maybe @dagar can help me on this ? As I've seen he wrote this script at first.

DonLakeFlyer, 2019-07-30:
Sorry, I don't know much about this at all to help. I'm not much of a Linux person.

DonLakeFlyer, 2019-07-30:
> I just rebuilt stable 3.5 and the app image worked for me when I ran it.

So it's unclear to me if the AppImage which has been released for Stable 3.5 is broken or not?

tilaktilak, 2019-07-30:
It's also working for me. But for instance libts is meant to be used for touchscreen i guess, so not sure if my ubuntu  is a completely good test config.

DonLakeFlyer, 2019-08-13:
Ah, I see. So current builds are likely ok on non-touchscreen systems which I assume are the majority case. Still good to get this all fixed up though.

pestophagous, 2020-12-15:
This script is still "broken" in the (IMO accurate) sense that @tilaktilak described in 2019.

In a broader sense the script is also "working great" because it does seem to produce AppImage(s) that everyone uses without hassle.

Three of the URL(s) hardcoded in the script are guaranteed to 404 100% of the time.

The good news is: if 0% of your end users have noticed anything wrong with the AppImage(s), then I think you can quite likely just remove the 3 offending `wget` usages from the script entirely.

Here are the three that 404:

```
http://ftp.us.debian.org/debian/pool/main/u/udev/udev_175-7.2_amd64.deb
http://ftp.us.debian.org/debian/pool/main/s/speech-dispatcher/speech-dispatcher_0.8.8-1_amd64.deb
http://ftp.us.debian.org/debian/pool/main/t/tslib/libts-0.0-0_1.0-11_amd64.deb
```

I also recommend (and myself use religiously) the following line at the top of all my bash scripts:

```
#!/bin/bash

set -Eeuxo pipefail # https://vaneyckt.io/posts/safer_bash_scripts_with_set_euxo_pipefail/
```

Which would have caused the 404 errors to surface in your CI as soon as the URL(s) went offline.

-----------------------------

To (possibly) convince yourself that it isn't even necessary to `wget` these items into the AppImage:

1. Download https://github.com/mavlink/qgroundcontrol/releases/download/v4.0.11/QGroundControl.AppImage
2. `mkdir mount_here && sudo mount -o loop v4.0.11_QGroundControl.AppImage mount_here`
3. `find mount_here/ -iname '*udev*'`
4. `find mount_here/ -iname '*speech*disp*'`
5. `find mount_here/ -iname '*libts*'`

If you do those steps, you will see that none of the 3 items are included anywhere in the AppImage.

So if your users aren't complaining, then either:

- Nobody is using the linux AppImage, or
- Nobody is using the newer AppImages (for almost 2 years?), or
- Those who use the AppImage don't rely on whatever plugin-based features might require those 3 items, or
- Those who use the AppImage can be relied on to "bring their own libts" (and other dependencies)

In any case, the script is absolutely failing to include those items into the AppImage. And you can see the proof in the AppImage(s) available for download.

So it cannot possibly make things worse to edit the offending lines in `create_linux_appimage.sh`.

closed 2024-06-02 by HTRamsey