mavlink/qgroundcontrol #7639: Map tiles not updating/loading on QGC 3.5.4

laurent1784, 2019-07-29:
## Expected Behavior
Map tiles shall update when connected to the internet via WiFi on QGC 3.5.4 in order to perform missions.

## Current Behavior
Map tiles are not updating on QGC 3.5.4. I am not able to plan or program any mission since maps are not showing.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1.
2.
3.
4.

## System Information
When posting bug reports, include the following information
- Operating System: Android KitKat on tablet.
- QGC Version: 3.5.4
- QGC build: stable
- Flight Controller: pixhawk
- Autopilot (with version): arducopter 3.6.9

## Detailed Description
Not able to update maps tiles on QGC. Get an error message stating no internet connection but wifi connection is working fine with chrome and all other apps.

## Log Files and Screenshots
- 
[image]
[image]

DonLakeFlyer, 2019-07-29:
@dogmaphobic What's the "No internet access" determining factor?

dogmaphobic, 2019-07-29:
That's on Android only. It tests Internet by testing Google's DNS (8.8.8.8:53)

dogmaphobic, 2019-07-29:
The notes in the code:
```
        To test if you have Internet connection, the code tests a connection to
        8.8.8.8:53 (google DNS). It appears that some routers are now blocking TCP
        connections to port 53. So instead, we use a TCP connection to "github.com"
        (80). On exit, if the look up for “github.com” is under way, a call to abort
        the lookup is made. This abort call on Android has no effect, and the code
        blocks for a full minute. So to work around the issue, we continue a direct
        TCP connection to 8.8.8.8:53 on Android and do the lookup/connect on the
        other platforms.
```

laurent1784, 2019-07-29:
so do i have to set up something on the android tablet?  I do have a vpn and dns setup on my wifi router?  Shall i remove the dns and see if that is the issue?

DonLakeFlyer, 2019-07-29:
> I do have a vpn and dns setup on my wifi router.

Guess would be something there causing the problem to not be able to reach Google DNS.

DonLakeFlyer, 2019-07-29:
@dogmaphobic I wonder if there should be a backdoor setting somewhere you can turn which says just skip the internet check and assume it's available.

dogmaphobic, 2019-07-29:
Not that hard to add. A new setting under General Settings? (yuk)

DonLakeFlyer, 2019-07-29:
Yeah, I know gross. But it doesn't seem like we hit something quirky here every once in a while.

Then maybe later some sort of overhaul of General Settings page is order with and advanced or something.

dogmaphobic, 2019-07-29:
I guess it's time to split General Settings into multiple panels as I did with Joystick Setup. I will work on that later this week...

laurent1784, 2019-07-30:
Hi guys. thanks for the help.  Curiously enough, I tried using my smartphone as a hotspot and connected the tablet to it via wifi.  Maps now load perfectly.  I also removed the vpn from my router but still had issues.  I guess it has to be something with the wifi routers or that they might have a blocked port from factory.  Can it be maybe a port issue on the router or a firewall?

PR #7660 (2019-08-05): Make Internet test optional.
In response to #7639 (and many others), this makes the test for Internet before accessing Internet resources optional. It is set to true by default (do test Internet connection).

When disabled, it simply ignores the tests and always assume Internet is available. The tests are done to avoid countless timeouts while downloading map tiles, etc. If you really don't have Internet, you will see a great deal of map tile download errors accordingly.

This is specially important for folks running Android devices in China, where most everything is blocked and the tests alwa [... cut]
closed 2019-08-05 by dogmaphobic