mavlink/qgroundcontrol #7642: master does not show action buttons

FlavioTonelli, 2019-07-30:
Current master 4b15d5bcf703ae9fa6ee0580e51871fdf976ad62 does not correctly show the action buttons(see screenshot)
[image]

DonLakeFlyer, 2019-07-30:
I think there is a disabled color theme problem where they are black on black.

FlavioTonelli, 2019-07-30:
I can confirm they correctly show when active.

DonLakeFlyer, 2019-07-30:
Also not sure why there is all that blank space between the toolbar and the toolstrip.

dogmaphobic, 2019-07-30:
Stefan was working on the tool strip. Let me find out what's going on...

FlavioTonelli, 2019-10-16:
I guess this has been fixed in the meantime. I can't reproduce anymore on master.

closed 2019-10-16 by FlavioTonelli