mavlink/qgroundcontrol #7656: Error message interrupts firmware upgrade

dwiel, 2019-08-01:
## Expected Behavior
When I try to select a firmware from the dropdown from the firmware page, I shouldn't get interrupted even if `One or more vehicle components require setup prior to flight.`

## Current Behavior
While I am selecting which firmware I want, a yellow orange warning pops up `One or more vehicle components require setup prior to flight.` and forces me back to the summary page. Once that is happened, when I go back to the firmware page it asks me to `unplug the pixhawk and/or Radio from USB.` Unplugging and plugging back starts me back at the beginning with another warning pop up appearing soon after. I could potentially start the firmware upgrade quickly before the pop up comes, but i am worried i'm going to break my flight controller if the pop up aborts the flashing process prematurely.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Leave the radio unconfigured
2. Navigate to firmware page
3. Plug in flight controller over USB
4. Wait a few seconds for a connection to be made and the warning message to appear, pulling you away from the firmware page back to the summary page.

## System Information
When posting bug reports, include the following information
- Operating System: macOS 10.14.6
- QGC Version: 3.5.4
- QGC build: stable
- Flight Controller: mRo-X2.1 Autopilot
- Autopilot (with version): ArduPlane V3.9.8 (6ea22c9c)

## Log Files and Screenshots
[image]

FlavioTonelli, 2019-08-01:
Are you powering the board through an external power source?
During FW upgrade it shall be powered by usb only.

dwiel, 2019-08-01:
I am only powering it via USB. I did find out that I could avoid this issue by holding down the safety switch as I plugged it into USB. However, it definitely took me quite a while to figure this out. Would have been nice to not have to do that.

dwiel, 2019-08-01:
btw, this was a brand new board never before used and so it was correct that the RC had never been calibrated.

DonLakeFlyer, 2019-08-13:
This is because the board won't stick in bootloader mode and blows by that and does a full boot. Turn on FirmwareUpgradeLog (https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html) for more details.

dwiel, 2019-08-14:
I only have this problem if there is an error message though, otherwise it's not an issue.

DonLakeFlyer, 2019-08-14:
My guess is that is a bit of a fluke with respect to the problem. That message is shown by QGC based on various parameter settings which means the firmware itself knows nothing about it. The fundamental problem is that the board is not staying in bootloader mode. The details from FirmwareUpgradeLog would tell me more.

closed 2026-10-04 by github-actions[bot]