mavlink/qgroundcontrol #7659: Need help with adding Here+ RTK support for Android

myeabhishek, 2019-08-02:
Hello Team,

I want to use Here+ RTK with qgroundcontrol android application. Please tell me how I can go about it? 

I can also add the support myself but I would love if someone can point me in the right direction.

DonLakeFlyer, 2019-08-13:
Not sure if it's possible given the way serial port code works in Android. Just follow the RTK code which is #ifdef'ed out in mobile builds.

wattsinnovations, 2020-05-08:
Hello has there been any update with this since August? We would potentially be very interested in assisting with this as well. Thanks!

DonLakeFlyer, 2020-05-08:
Someone was trying to get it to work but didn't succeed. Not sure where they ran into problems.

DonLakeFlyer, 2020-05-08:
WIP was here: https://github.com/mavlink/qgroundcontrol/pull/8210

DonLakeFlyer, 2020-05-12:
Not that I know of

HTRamsey, 2024-06-02:
Android serial driver is being updated

closed 2024-06-02 by HTRamsey